ob-f/OpenBot #2: Tips on printing on a printer with smaller build volume?

katnoria, 2020-08-27:
First of all thank you for sharing this project. I am really excited to make one and run my on models on it.
Do you have any tips about printing on a smaller build volume? I have Ultimaker 2 extended which has a build volume of 223 mm × 223 mm × 305 mm. The body of the bot does not fit? The best I can get is this 
[image]

thias15, 2020-08-27:
Hi @katnoria 

I have made a slim version of the model that fits into a 220x220 build surface. Please try it and let me know. 
If this works, I can also create a matching top cover. 

[body_bottom_slim.slt](https://github.com/intel-isl/OpenBot/blob/master/body/body_bottom_slim.stl)

[image]

katnoria, 2020-08-28:
hi @thias15 

Thank you for making the slim version. It work 👍 
[image]

Would you please create the slim version of top cover too?

VCU-RVI-dataset, 2020-08-28:
hi @thias15

My printer is 200X200X200mm. Would you mind uploading the editable part files as well? If so, we can shrink the size or split the base and top into several pieces on our own.

thias15, 2020-08-29:
I made a slim version for the top: [body_top_slim.stl](https://github.com/intel-isl/OpenBot/blob/master/body/body_top_slim.stl). 
[image]

thias15, 2020-08-29:
@VCU-RVI-dataset, I have also added the STEP files for the top and bottom of the chassis for more flexibility (see updated README). Please let me know if you figure out a good solution for smaller build volumes.

closed 2020-08-29 by thias15

katnoria, 2020-08-30:
Thank you @thias15