ob-f/OpenBot #13: Wiring diagram

Tmin10, 2020-09-01:
May you provide wiring diagram for the robot? 

I'm little bit confused with point 17 in the build instructions:
- (Optional) Connect the voltage divider to pin A7 of the Arduino

What is the voltage divider, why we need it and how we can connect it to the robot.

Other question about point 14:
Why we need connect echo and trigger pins together in one arduino port? Other samples told us to use two ports for this...

P.S. Thank you for your project!

thias15, 2020-09-01:
The voltage divider is used to measure the battery voltage. The battery has a voltage of 9.6-12.6V. But the ADC of the Arduino can only measure a voltage of up to 5V. For example, you can use a 20k and 10k resistor and then set the multiplier in the app to 3. One end of the 20k resistor is connected to Vin and the other to the Arduino pin A7 and one end of the 10k resistor. The other end of the 10k resistor is connected to GND. More info [here](https://ohmslawcalculator.com/voltage-divider-calculator).
[image]

We use the NewPing library for the Ultrasonic sensor. It supports single pin operation and automatically toggles this pin to be an input and output depending if the echo signal is sent or it is waiting for the trigger signal to be received. You can also connect it to two pins as in other tutorials if you prefer. Just make sure to keep it event-driven as in the code, otherwise the ultrasonic sensor will block the rest of the code.

Tmin10, 2020-09-01:
Got it. So voltage divider is about arduino power supply. 
Maybe it's better to use some DC-DC convertor since we losing part of the energy as heating on resistors

Tmin10, 2020-09-01:
Hm, I reread instructions again and looks like this voltage divider isn't for power supply, sorry.
Is it for battery charge level checking? I confused because there is no resistors in the bill of materials...

thias15, 2020-09-02:
Yes, it is to check the battery voltage and it is optional.

thias15, 2020-09-04:
Closing due to inactivity.

closed 2020-09-04 by thias15

Tmin10, 2020-09-04:
I'm sorry but my request still open. Maybe in future you can add picture with the wiring diagram? 
In other issue I found question about wiring motors. It could help with robot assembly for others.

thias15, 2020-09-04:
I have added a wiring diagram. Does this answer your questions? If not feel free to reopen the issue or open another one.

Tmin10, 2020-09-04:
Cool, thank you! Sorry, I missed your commit.