ob-f/OpenBot #15: Wrong camera in the app

Tmin10, 2020-09-02:
Hello. I run the app on my phone but it display picture from front camera. After debugging I realize that I need to change https://github.com/intel-isl/OpenBot/blob/master/android/app/src/main/java/org/openbot/CameraActivity.java#L110 to 0 (LENS_FACING_FRONT) and it working fine... I don't know why front and back are reversed in my case but it could help you in future for debug the same problems.

My device is Asus Zenfone Max Pro (M2) ZB631KL, Android 9. Other cameras apps working fine.

thias15, 2020-09-02:
We have not encountered this on any phone before. Thanks for opening this issue, we will keep an eye on it. I will close it for now, since I don't expect this to be a general issue.

Tmin10, 2020-09-02:
Yep, thanks. Maybe in future it is possible to add camera switch into the settings.

eagleanurag, 2020-10-11:
> Hello. I run the app on my phone but it display picture from front camera. After debugging I realize that I need to change [...]

tin my case also when i used my old Samsung J7 Max by default front camera opens

vkuehn, 2020-11-17:
Hi I have a solution for that, can you kindly assign it to me ?
Sorry it will be not fully compliant with your naming rules..this time

closed 2020-11-19 by thias15

vkuehn, 2020-12-08:
is there an issue why it is a ToDo again ?