ob-f/OpenBot #16: Tensorflow-lite not found error?

Apurv505, 2020-09-02:
Hi,
First of all thanks for making such an awesome robot!!! I have been waiting for something like this for years!!!
I tried to follow the app build instructions but hit an error in the middle.

**It says tensorflow-lite doesn't exists in the folder**. I am an amateur in android studio so don't really know if this should happen or not. I checked the cloned folder and there  wan't any tensor flow file. 
Hope this is not confusing and I would really appreciate help on this.

Thank you!!!

_Download https://services.gradle.org/distributions/gradle-5.4.1-all.zip (132.77 MB)
Download https://services.gradle.org/distributions/gradle-5.4.1-all.zip finished succeeded, took 7 m 49 s 218 ms
Starting Gradle Daemon...
Gradle Daemon started in 1 s 617 ms
**<ij_msg_gr>Project resolve errors<ij_msg_gr><ij_nav>/home/pinkpanther/Github_cloned/OpenBot/android/build.gradle<ij_nav><i><b>root project 'android': Unable to resolve additional project configuration.</b><eol>Details: org.apache.tools.ant.BuildException: Basedir /home/pinkpanther/Github_cloned/OpenBot/android/tensorflow-lite does not exist</i>**
File /home/pinkpanther/.android/repositories.cfg could not be loaded.
Checking the license for package Android SDK Build-Tools 28.0.3 in /home/pinkpanther/Android/Sdk/licenses
License for package Android SDK Build-Tools 28.0.3 accepted.
Preparing "Install Android SDK Build-Tools 28.0.3 (revision: 28.0.3)".
"Install Android SDK Build-Tools 28.0.3 (revision: 28.0.3)" ready.
Installing Android SDK Build-Tools 28.0.3 in /home/pinkpanther/Android/Sdk/build-tools/28.0.3
"Install Android SDK Build-Tools 28.0.3 (revision: 28.0.3)" complete.
"Install Android SDK Build-Tools 28.0.3 (revision: 28.0.3)" finished.
Checking the license for package Android SDK Platform 28 in /home/pinkpanther/Android/Sdk/licenses
License for package Android SDK Platform 28 accepted.
Preparing "Install Android SDK Platform 28 (revision: 6)".
"Install Android SDK Platform 28 (revision: 6)" ready.
Installing Android SDK Platform 28 in /home/pinkpanther/Android/Sdk/platforms/android-28
"Install Android SDK Platform 28 (revision: 6)" complete.
"Install Android SDK Platform 28 (revision: 6)" finished.
Calling mockable JAR artifact transform to create file: /home/pinkpanther/.gradle/caches/transforms-2/files-2.1/60b701dab9dbd0c40a2aad2e3a66fedc/android.jar with input /home/pinkpanther/Android/Sdk/platforms/android-28/android.jar

Deprecated Gradle features were used in this build, making it incompatible with Gradle 6.0.
Use '--warning-mode all' to show the individual deprecation warnings.
See https://docs.gradle.org/5.4.1/userguide/command_line_interface.html#sec:command_line_warnings

CONFIGURE SUCCESSFUL in 28m 46s_

Tmin10, 2020-09-02:
Hm, I successfully build the app but I also have no tensorflow-lite folder...
My build log:
```
Executing tasks: [:app:assembleDebug] in project C:\...\OpenBot\android

WARNING: The specified Android SDK Build Tools version (28.0.3) is ignored, as it is below the minimum supported version (29.0.2) for Android Gradle Plugin 4.0.1.
Android SDK Build Tools 29.0.2 will be used.
To suppress this warning, remove "buildToolsVersion '28.0.3'" from your build.gradle file, as each version of the Android Gradle Plugin now has a default version of the build tools.
> Task :app:preBuild UP-TO-DATE
> Task :app:preDebugBuild UP-TO-DATE
> Task :app:compileDebugAidl NO-SOURCE
> Task :app:generateDebugBuildConfig UP-TO-DATE
> Task :app:compileDebugRenderscript NO-SOURCE
> Task :app:javaPreCompileDebug UP-TO-DATE
> Task :app:generateDebugResValues UP-TO-DATE
> Task :app:generateDebugResources UP-TO-DATE
> Task :app:mergeDebugResources UP-TO-DATE
> Task :app:createDebugCompatibleScreenManifests UP-TO-DATE
> Task :app:extractDeepLinksDebug UP-TO-DATE
> Task :app:processDebugManifest UP-TO-DATE
> Task :app:processDebugResources UP-TO-DATE
> Task :app:compileDebugJavaWithJavac UP-TO-DATE
> Task :app:compileDebugSources UP-TO-DATE
> Task :app:mergeDebugShaders UP-TO-DATE
> Task :app:compileDebugShaders NO-SOURCE
> Task :app:generateDebugAssets UP-TO-DATE
> Task :app:mergeDebugAssets UP-TO-DATE
> Task :app:processDebugJavaRes NO-SOURCE
> Task :app:mergeDebugJavaResource UP-TO-DATE
> Task :app:checkDebugDuplicateClasses UP-TO-DATE
> Task :app:mergeLibDexDebug UP-TO-DATE
> Task :app:dexBuilderDebug UP-TO-DATE
> Task :app:desugarDebugFileDependencies UP-TO-DATE
> Task :app:mergeExtDexDebug UP-TO-DATE
> Task :app:mergeProjectDexDebug UP-TO-DATE
> Task :app:mergeDebugJniLibFolders UP-TO-DATE
> Task :app:validateSigningDebug UP-TO-DATE
> Task :app:mergeDebugNativeLibs UP-TO-DATE

> Task :app:stripDebugDebugSymbols UP-TO-DATE
WARNING: Compatible side by side NDK version was not found. Default is 21.0.6113669.
Compatible side by side NDK version was not found. Default is 21.0.6113669.

> Task :app:packageDebug
> Task :app:assembleDebug

BUILD SUCCESSFUL in 3s
23 actionable tasks: 1 executed, 22 up-to-date

Build Analyzer results available
```

thias15, 2020-09-02:
So is it working now? 
You need to run the gradle sync and then compile.
The tensorflow-lite folder will be generated automatically.

closed 2020-09-03 by thias15

Apurv505, 2020-09-04:
Yes! Thank you so much!!
The gradle sync completed and when I rebuilt it, it was successful. Now I am waiting for speed sensors to arrive and I'll start making the bot.
Thank you so much for helping!!!