ob-f/OpenBot #17: 如何获取 3D 机身

mr-shitou, 2020-09-02:
@leegang 
嗨，我想组装一个，我看其他的零配件都有写店铺地址，但是 3D 打印的机身没有说具体的怎么去弄
对于这方面不是特别的了解，想请教下怎么在国内并且没有相关设备的情况下去弄一个机身，以及大概的价格是多少
谢谢

thias15, 2020-09-02:
如果您没有3D打印机，则可以使用在线服务。 您将上传stl文件，它们将打印模型并将其运送给您。 我不知道如何在中国找到这样的服务，但是我确定它存在。

@leegang 你能帮我吗？

leegang, 2020-09-02:
> 如果您没有3D打印机，则可以使用在线服务。 您将上传stl文件，它们将打印模型并将其运送给您。 我不知道如何在中国找到这样的服务，但是我确定它存在。
> 
> 
> 
> @leegang 你能帮我吗？

有易立创、云铸 、魔猴 等在线打印平台。
@mr-shitou 具体价格要上传模型文件才能估价，明天我看看。

mr-shitou, 2020-09-03:
我在 TB 有找到打印店铺，但是有打印要求，不知道能不能满足机身的要求。还有材料、制作方法、工艺等，我就懵了😵
希望能详细些，thanks a lot all of you

leegang, 2020-09-03:
> 我在 TB 有找到打印店铺，但是有打印要求，不知道能不能满足机身的要求。还有材料、制作方法、工艺等，我就懵了😵
> 希望能详细些，thanks a lot all of you

[image]

timaker, 2020-09-03:
我找到一家提供整套配件的卖家，搜索openbot就找到，好像店铺名叫钛创客

mr-shitou, 2020-09-03:
@leegang 能给个链接么，我找了几个都比你的贵，3q

leegang, 2020-09-03:
> @leegang 能给个链接么，我找了几个都比你的贵，3q

https://www.qsy3d.com/quote.html

thias15, 2020-09-03:
I believe the issue is resolved.

closed 2020-09-03 by thias15