ob-f/OpenBot #21: Which motors connect to which output?

mikeh9, 2020-09-04:
Really appreciate your project and all the hard work you put in. I know the wiring diagram is coming, but can you comment on which motors goes to Out1/2 and which motors go to Out3/4?

I'm writing this to help others out as well. :-)

thias15, 2020-09-04:
Insert the positive and negative leads of the two **left** motors into OUT1 (+) and OUT2 (-) of the L298N board

Insert the positive and negative leads of the two **right** motors into OUT4 (+) and OUT3 (-) of the L298N board

The reason that +/- is swapped for left and right is that the motors are mounted with 180 degrees rotation, so they turn in the opposite direction. 

The leads of the motors on the same side are connected, since they are controlled by the same H-bridge. I recommend to solder the wires of the same polarity for both motors on the same side. This way it's easier to install them and make sure there is a good connection.

moosa1996, 2020-09-04:
Hi I have a issue on connecting my nano USB to my mobile type c. cant able to establish USB Connection.I have enabled my developer USB debugging in my android mobile. But still device is not connected in OpenBot app.What is the issue ?

mikeh9, 2020-09-04:
Your reasoning is correct, but I think you mistyped what you meant. I think you wanted to write:

Insert the positive and negative leads of the two left motors into OUT1 (+) and OUT2 (-) of the L298N board

Insert the negative and positive leads of the two right motors into OUT3 (+) and OUT3 (-) of the L298N board

With this configuration my car runs properly. Now onto the training. :-)

closed 2020-09-04 by mikeh9