ob-f/OpenBot #22: Openbot Android app

moosa1996, 2020-09-04:
Hi I have a issue on connecting my nano USB to my mobile type c. cant able to establish USB Connection.I have enabled my developer USB debugging in my android mobile. But still device is not connected in OpenBot app.What is the issue ?

francisduvivier, 2020-09-04:
Which phone do you use? It might not support USB OTG

thias15, 2020-09-04:
When you first connect the Arduino to your phone it will ask you for permission to access the USB device. Did you see this dialog? 

If **yes**, you need to give permission. 

If **no**, you probably have a connection issue. 
- Double-check the baud rate of App and Arduino are the same (they should be unless you changed something. 
- Try another cable. 
- Confirm your phone supports USB-OTG.

mikeh9, 2020-09-04:
Can you confirm you used the cable specified in the BOM? It needs to be an OTG cable (ID connected to ground)

moosa1996, 2020-09-05:
> Which phone do you use? It might not support USB OTG
> […](#)

I'm using One plus 5T which supports OTG

moosa1996, 2020-09-05:
> When you first connect the Arduino to your phone it will ask you for permission to access the USB device. Did you see this dialog?
>
> If **yes**, you need to give permission.
>
> If **no**, you probably [...]

Yes it asked for permission to access the USB device. I gave yes and the openbot app opens automatically when my arduino nano is plugged into OTG. but when the app is opened it shows a dialogue text "Please Check the USB Connection". Why so?

moosa1996, 2020-09-05:
> Can you confirm you used the cable specified in the BOM? It needs to be an OTG cable (ID connected to ground)

I used 2 type of cable converter. One is "Type C to USB female" and other is "Mini to USB male" i have attached my converter cables for your reference 
[image]

thias15, 2020-09-05:
Should work in principle, but we never tried with adapters. Can you verify with another phone or device that the cables work?

thias15, 2020-09-07:
What happens when you try to toggle the connection switch in the app? Are you sure the Arduino has the firmware flashed correctly? Can you try to flash it again and then open the serial monitor to see if the Arduino is connecting and sending it's messages.

thias15, 2020-09-09:
Closing due to inactivity.

closed 2020-09-09 by thias15