ob-f/OpenBot #23: BOM Question, Replacement for TMCP1D104MTRF?

ssurovich, 2020-09-06:
Hi team - First, great project - I cant thank you enough for the work the team has done, Im very excited to play around with the project.

On the BOM, you have a Capacitor, Vendor part TMCP1D104MTRF, but at LCSC it has been discontinued, and the only direct replacement has also been discontinued.  They have every other part, so I was wondering if you had a replacement capacitor from LCSC that you could suggest?   The closest match I could find is TAJA104K035RNJ from AVX - It may be a good replacement, but before ordering the entire BOM from LCSC, it would be great to have a confirmation.

Thanks!

thias15, 2020-09-06:
As long as it has the same package it should work fine. Can you post the link?

thias15, 2020-09-06:
I guess it is this one. 
https://lcsc.com/product-detail/Tantalum-Capacitors_AVX_TAJA104K035RNJ_100nF-104-10-35V_C7173.html

The package is a bit bigger, but you can probably get away with it since the pads are quite big. May be a bit harder to solder on.

ssurovich, 2020-09-07:
Hey @thias15 - Thanks, that was my bigger question, the size for the PCB - but theres always some "method" to make it work (one way or another) - I appreciate the quick reply and suggestion - Thanks again!

thias15, 2020-09-07:
Closing the issue for now. Would be great if you can post how it worked out after you get the components.