ob-f/OpenBot #24: what are the robot's dimensions?

maraoz, 2020-09-08:
I couldn't find it in the site or README, and it's not clear to me from the models, since I'm new to 3d printing.
Thanks!

ps: any chance you have body parts that fit in a 200x200 build plate?

thias15, 2020-09-08:
The required build plate dimensions for the slim body are 223x223. I'm afraid that it will be tricky to fit into a 200x200 build plate. However, we also provide the step models. So you could adapt the model, maybe split top and bottom into two pieces and then assemble them.

thias15, 2020-09-09:
Closing due to inactivity.

closed 2020-09-09 by thias15

maraoz, 2020-09-09:
I'd still like to know what the robot's dimensions are. Where can I find that info? Thanks

manair, 2020-09-17:
If anyone is able to split this for a 200x200 Printer 3d printer plate, and reassemble it successfully please share the .stl file. Thank you