ob-f/OpenBot #26: Clarification needed regarding the BOM and U1 Quantity

mondobongo64, 2020-09-11:
Please confirm the quantity for ID 7 in the BOM.csv. The BOM states 2 but this seems to be in contradiction to the written description of the Electrical Design contained in the Paper. 

ID | Name | Designator | Footprint | Quantity | Manufacturer Part | Manufacturer | Supplier | Supplier Part | LCSC Assembly
7 | ARDUINO_NANO | U1 | ARDUINO_NANO | 2 | 2685Y-115CNG1SNA01 | HOAUC | LCSC | C350309 |  

Thanks

thias15, 2020-09-14:
This is the pin header for the Arduino. You need two rows of 15 pin headers, so the BOM is correct.

I do not think this contradicts with the paper. The description in the paper is about the circuit with DIY components without the PCB. Could you maybe point out the section in the paper where you spotted the contradiction?

closed 2020-09-14 by thias15

agios, 2020-09-15:
Hi. I'm not an expert, but should the second pin header be given another designation and also added to the centroid file file for assembly?