ob-f/OpenBot #32: I can not put the customer PCB and L298N into the box at the same time.

raygong1207, 2020-09-15:
1: I can not put the customer PCB and L298N into the box at the same time, Because no enough space.
2: It seams that the L298N and customer PCB (with the chip DRV8871DDAR SOP8) both for driver the motor；

So i guess that just need one of  L298N and customer PCB ,right? If not. please tell me the different function for L298N and customer PCB.

3: I find something of customer PCB not match with the instruction below:
"14,Connect Echo and Trigger of the ultrasonic sensor to pin D4 of the Arduino.". 
In the fact, the customer PCB is connect the Connect Echo and Trigger of the ultrasonic sensor to pin D3.

So, i just do not need customer PCB,right?

thias15, 2020-09-15:
You cannot use both at the same time. If you use the PCB, you don't need the L298N. The PCB has motor driver chips, voltage divider circuit, etc. integrated. Motors and sensors can just be connected with Dupont cables. It is meant for building multiple robots quickly. The PCB pinout is slightly different; in the Arduino code, you need to select either DIY or PCB.

thias15, 2020-09-15:
If you follow the DIY build instructions, you will need to do the wiring manually and you do not need the PCB. Does this answer your questions?

thias15, 2020-09-16:
Updated README to make this more clear.

closed 2020-09-16 by thias15

raygong1207, 2020-09-16:
Thanks a lot! Your answer is very clear. I got it.