ob-f/OpenBot #34: the PCB has an error.Arduino nano d4 can not set interrupt.

KevinBuddy, 2020-09-16:


thias15, 2020-09-16:
Yes, we have already updated the PCB. But are still waiting for prototypes for testing. 
If you have already ordered the old PCB, you can initialize the following variables
```
int temp = 0;
int speed_l = 0;
int speed_r = 0;
```
disable the interrupts
```
  //attachInterrupt(digitalPinToInterrupt(PIN_SPEED_L), speed_left, RISING);
  //attachInterrupt(digitalPinToInterrupt(PIN_SPEED_R), speed_right, RISING);
```
and use this code snippet in the the main loop to read the speed sensors manually:
```
  temp = digitalRead(PIN_SPEED_L);
  if (temp != speed_l) {
    speed_l=temp;
    speed_left();
  }
  temp = digitalRead(PIN_SPEED_R);
  if (temp != speed_r) {
    speed_r=temp;
    speed_right();
  }
```

Note that one revolution will correspond to 40 ticks now, since every change (RISING + FALLING) is detected.

KevinBuddy, 2020-09-17:
ok, thank you.

closed 2020-09-17 by thias15

thias15, 2020-09-30:
This is addressed in the code now. Simply set the PCB version. No need to add the code above.

KevinBuddy, 2020-09-30:
Thank you.I have a question,in autopilot mode, where is the cmd come from?