ob-f/OpenBot #41: the slim version doesn't fit on plate 223x223

ronki2304, 2020-09-18:
Hello I tried to print the slim version

but body_top_slim.stl doesn't fit

I used an ultimaker 2 which have a plate of 223*223mm

thias15, 2020-09-18:
Did you rotate it by 45 degrees?

ronki2304, 2020-09-18:
Yes I have rotated it 

the bottom filled  perfectely but not the top

[image]

I see your PR, I don't know what is the difference, but the model doesn't fit on cura

thias15, 2020-09-18:
In the bottom left, the dimensions show as 223x223. Is there maybe some setting in Cura?

thias15, 2020-09-19:
Any luck?

ronki2304, 2020-09-21:
Hello sorry for my late answer,

what do you mean by some settings ???? those dimensions are the piece one on the build plate

thias15, 2020-09-21:
Well, you have an ultimaker 2 which as a build plate of 223x223. The model is also 223x223, so it should fit. So you probably have some setting enabled which prevents you from using the complete build area. 

- Have you selected build plate adhesion with a brim or a skirt? 
This will increase the overall size of your print.
To disable it, set *Build Plate Adhesion Type* to "None". 

[image]

Other settings that are probably not necessary, but may help to fit the model:
- Disable the second extruder 
- Disable prime blob (under build plate adhesion)

ronki2304, 2020-09-21:
Hello,
the build plate adhesion is already set to off and I don't have a second extruder,

i tried to change the g-code flavor but I don't see any change. i don't which setup has been made by @katnoria in [PR#2](https://github.com/intel-isl/OpenBot/issues/2)

thias15, 2020-09-21:
@ronki2304 Please try https://github.com/intel-isl/OpenBot/blob/develop/body/slim_body/slim_body_top.stl from the development branch. If this does not work, checkout the glueable model.

ronki2304, 2020-09-21:
OK it works thank you very much

closed 2020-09-21 by thias15