ob-f/OpenBot #43: Is there a way to stop the epochs, but not lose the generate tflite data?

chilipeppr, 2020-09-18:
Question. Can I stop the epochs run in the middle, but not lose the generating of the best/last tflite files? I'm in a situation where I trained for 100 epochs, but at about epoch 47 I reached the max training and now it's overfitting (I believe). I've been processing for like 7 hours so far, so to wait another 3 hours is a waste of time and energy, but I don't want to lose the work it took to get to the 47th epoch.

Here are my graphs of the output as it's going along. I'm using Tensorboard to show the graphs as the epochs are running.

[image]
[image]

thias15, 2020-09-18:
You can stop the training, but you'll loose the history. You can still load any of the saved epochs manually and convert them to tflite files.

thias15, 2020-09-18:
For example if you want the tf-lite file for checkpoint number 40, run the code below.
Note that after stopping the kernel, you will either have to run the cells that generate the checkpoint path or specify it manually. 
```
import utils
best_checkpoint = "cp-0040.ckpt"
my_tflite = utils.generate_tflite(checkpoint_path, best_checkpoint)
utils.save_tflite (best_tflite, checkpoint_path, "best")
```

chilipeppr, 2020-09-18:
Worked like a charm. Here's my final code that I added as a cell and will comment it out on a normal run, or uncomment it later to manually generate.

```python
import utils
checkpoint_path = "./models/thu9PMRun_pilot_net_lr0.0001_bz128_bn/checkpoints"
best_checkpoint = "cp-0044.ckpt"
my_tflite = utils.generate_tflite(checkpoint_path, best_checkpoint)
utils.save_tflite(my_tflite, checkpoint_path, "best")
print("Saved out best.tflite. checkpoint_path:", checkpoint_path, "best_checkpoint:", best_checkpoint)
```

closed 2020-09-19 by thias15