ob-f/OpenBot #44: 100% Successful AI Run!

chilipeppr, 2020-09-19:
I realize this is not an "issue", so this is not the best place to post this, but since there's no discussion forum for this project it's the best place to share this as I may be the 1st person to get a 100% successful AI training outside of the author of this project, so I'd like to inspire others to get to this spot as well.

https://www.youtube.com/watch?v=vfas-lWZl4c

Here's the video of the OpenBot making it around my kitchen with just the Autopilot turned on. This was an amazing moment for me as it took a couple weeks to get to this spot. I trained about 120,000 images with about 20,000 test images. I even had my kids help gather the training data since it got so tedious.

thias15, 2020-09-19:
Hi @chilipeppr 

Thanks for sharing. Getting an AI model to work well is hard work, but you did it! Congratulations. 🎉

thias15, 2020-09-21:
There is now a [Slack](https://join.slack.com/t/openbot-community/shared_invite/zt-hpso8cfl-JO9OVhVMdUWvR4vDXwcMGA) channel for people to discuss the project, ask for help and share their successes and failures.

closed 2020-09-21 by thias15

zlite, 2020-11-10:
> There is now a [Slack](https://join.slack.com/t/openbot-community/shared_invite/zt-hpso8cfl-JO9OVhVMdUWvR4vDXwcMGA) channel for people to discuss the project, ask for help and share their successes and failures.

That Slack link is now expired. Can you post a new one?