ob-f/OpenBot #45: 有了PCB版，和组件的接线是什么样的？

2like, 2020-09-19:
@zhaoyshine 
如题，能否补充下中文版readme。

zhaoyshine, 2020-09-19:
抱歉我按选项一来的，选项二这块我不太清楚，选项一走线我选的 杜邦线 母-母、公-母、公-公接头我各准备了十根，应该够用。目前我东西也还在路上，没有开始

2like, 2020-09-19:
等你选项一的好消息！
@zhaoyshine

thias15, 2020-09-20:
Is this issue resolved?

2like, 2020-09-21:
> Is this issue resolved?

Can you give a detailed description of how to use PCB

thias15, 2020-09-21:
- Attach Arduino Nano to the U1 header
- Connect motors to M1-M4
- Connect power source to Vin
- Connect sensors to the appropriate header pins

closed 2020-09-22 by thias15

2like, 2020-10-09:
> 抱歉我按选项一来的，选项二这块我不太清楚，选项一走线我选的 杜邦线 母-母、公-母、公-公接头我各准备了十根，应该够用。目前我东西也还在路上，没有开始

杜邦线怎么买？搞定了吗？

zhaoyshine, 2020-10-10:
早搞定了，杜邦线 母-母、公-母、公-公接头我各准备了十根，淘宝大把随便买。体验下来自动跟随人体识别有点不行，随便一件衣服就识别成人了，不能稳定跟某一个人，目前一直在玩手柄模式

zhaoyshine, 2020-10-10:
这个是我demo版时拍的照片，应该是对的
[image]