ob-f/OpenBot #46: Consider tagging "releases" to help us make sense of PCB and firmware revs

DiggidyDave, 2020-09-19:
I have v1 of the PCB and was getting ready to assemble with my daughter and see now that you have a V2 of the PCB and firmware has changed etc. It isn't clear to me which commit to checkout to have a working, coherent "v1" repository that should mostly just work if I follow directions in the READMEs.

thias15, 2020-09-20:
All the demos work with the firmware as it is. The PCB mainly has cosmetic changes, there are just two functional changes related to voltage monitoring and speed sensing. Neither are currently used the app. I did not get a chance yet to update the code for the Arduino. But it will have a flag for PCBv1 and PCBv2.

DiggidyDave, 2020-09-20:
Thanks for responding, @thias15 To be clear, I'm not talking about build flags etc, I'm talking about commit tagging in git. 

I think you have an amazing project here that likely has legs (1.4k stars already!).  I think you and the community around this will have fewer headaches over time with a sensible versioning and release tagging strategy sooner rather than later.  Imagine a future where you are on the 7th rev of the PCB and there are 1000 commits in source, and my arduino breaks... I just want to be able to checkout v1.0.0 (the release that I build my bot from, as opposed to 7.1.8 or whatever you happen to be on at that time) and have 100% confidence that the firmware and source will work. :-)

In any case, keep up the awesome work, this is a fantastic project. I'm hoping to complete my first build of it with my daughter today.

thias15, 2020-09-20:
@DiggidyDave Yes, I understand and there will be tagged releases in the future for sure. However, there are still a few small things I'm working on which I wanted to iron out before tagging the first release. I want to make sure at least a few people were able to complete all steps in the project. The challenge is also to find a good release strategy across the different components (e.g. people may have the old PCB, but still like new features in the firmware, the Android app, etc.) Ideally, each component could be tagged seperately, but I guess the only way to achieve this would be seperate repos and then add them as submodules. Then, it could be that releases of submodules will not be compatible with each other. I guess, in addition there could be releases across the repo with the submodules. If you have some suggestions, please let me know. As I mentioned, for now my plan was to add build flags to the Arduino code to support different PCBs.

DiggidyDave, 2020-09-20:
Agreed, I almost mentioned having independently versioned "projects" but that felt heavy for the current young state of the project. Cool, thanks for your responses! If I have any lightweight ideas/suggestions in the near term I'll pass them along.

closed 2020-09-20 by DiggidyDave

thias15, 2020-09-30:
Just for your info, we just released [v0.1.0](https://github.com/intel-isl/OpenBot/releases/tag/v0.1.0).