ob-f/OpenBot #47: Documentation for firmware needs lots of work

Cantfigurethisout, 2020-09-20:
I am new to this kind of thing, I realized that the firmware needed to get to the arduino board however if you read the documentation it mentions nothing.  I downloaded arduino IDE added the board in the board manager and sent the file to the board.  Got lots of errors in the IDE (don't know if it was something I did), tried to get the thing to drive....... nothing.  After multimeter testing terminals realized pins needed to be uncommented in the firmware.........sent it to the board with a large list of errors.  Still not getting any wheel spinning.  Scratching my head as to the issue.  I have bought everything to stick with the build as documented hoping I could have my handheld as build this.

thias15, 2020-09-20:
What error are you getting? You shouldn't need to comment anything.

thias15, 2020-09-20:
Did you install the NewPing library?

thias15, 2020-09-20:
Also let me give a general recommendation for issues in open-source projects.
1) Don't complain, be nice when asking for help. This is an open-source project. Nobody wants to help a complainer. 
2) Be specific about your error. Include screenshots, error logs, etc.
3) If you feel things could be done better, improve them and send a pull request.

crearo, 2020-09-20:
+1 for @thias15 's comment. This is actually one of the best documented hardware+software+3D-printing projects I've seen on GitHub!

Apart from writing out what exactly your current setup is, write about your problem specifically. 
Additionally, don't expect this project to have a start-from-scratch guide. If you're new, do you research on things you don't understand first.

Cantfigurethisout, 2020-09-20:
Didn't mean to be a complainer, I apologize for sounding that way, just trying to identify an area that was lacking (everything else has been fairly well documented, or at least it is mentioned what needs to be done).  

"You shouldn't need to comment anything."  Don't you have to delete some of the hashtag things to make it work?

#if (OPENBOT == DIY)
  #define PIN_PWM1 5
  #define PIN_PWM2 6
  #define PIN_PWM3 9
  #define PIN_PWM4 10
  #define PIN_SPEED_L 2
  #define PIN_SPEED_R 3
  #define PIN_VIN A7
  #define PIN_TRIGGER   4
  #define PIN_ECHO      4
  #define PIN_LED_RL 7
  #define PIN_LED_RR 8
#elif (OPENBOT == PCB)
  #define PIN_PWM1 10
  #define PIN_PWM2 9
  #define PIN_PWM3 5
  #define PIN_PWM4 6
  #define PIN_SPEED_L 2
  #define PIN_SPEED_R 4
  #define PIN_VIN A7
  #define PIN_TRIGGER   3
  #define PIN_ECHO      3
  #define PIN_LED_RL 7
  #define PIN_LED_RR 8
#endif

As far as errors go I got a lot, need to spend more time with it to make sure I am not doing anything dumb.  Not intending to ask for help yet on fixing my setup, like I said noticed an area that was lacking and wanted to point it out. 

If I was going to complain I would start with not being able to print on my ender 3 because the dimensions of the car are too big.  My solution for this sucked and I want to re-do it, if that is halfway decent I will post it so others with a smaller print volume can use it as well.

parixit, 2020-09-20:
Those aren't comments.  These are comments:
// ---------------------------------------------------------------------------
// This Arduino sketch accompanies the OpenBot Android application. 
//
// The sketch has the following functinonalities:
//  - receive control commands from Android application (USB serial)
//. - produce low-level controls (PWM) for the vehicle
//  - toggle left and right indicator signals 
//  - wheel odometry based on optical speed sensors
//  - estimate battery voltage via voltage divider
//  - estimate distance based on sonar sensor 
//  - send sensor readings to Android application (USB serial)
//
//  Dependencies (if sonar is required):
//  - NewPing library by Tim Eckel (Install via Tools --> Manage Libraries)
//  
// By Matthias Mueller, Intelligent Systems Lab, 2020
// ---------------------------------------------------------------------------

// PIN_PWM1,PIN_PWM2,PIN_PWM3,PIN_PWM4      Low-level control of left DC motors via PWM 
// PIN_SPEED_L, PIN_SPEED_R                  Measure left and right wheel speed
// PIN_VIN                                  Measure battery voltage via voltage divider
// PIN_TRIGGER                              Arduino pin tied to trigger pin on ultrasonic sensor.
// PIN_ECHO                                 Arduino pin tied to echo pin on ultrasonic sensor.
// MAX_DISTANCE                             Maximum distance we want to ping for (in centimeters). 
// PIN_LED_RL, PIN_LED_RR                   Toggle left and right rear LEDs (indicator signals) 

//------------------------------------------------------//
//DEFINITIONS
//------------------------------------------------------//

thias15, 2020-09-20:
@Cantfigurethisout What you posted are the pin definitions. You just need to set if you are using the DIY or the PCB version. That's all. 

If you are building the robot using the DIY instructions just leave as it is:
```
// Setup the OpenBot version
#define OPENBOT DIY
```

If you are using the PCB then change it to:
```
#define OPENBOT PCB
```

What is the size of the build plate on your 3D printer?

thias15, 2020-09-20:
@Cantfigurethisout I just merged a PR that adds a flag to disable the sonar. Try setting `HAS_SONAR 0`. This will remove compile errors related to not installing the NewPing library.

thias15, 2020-09-21:
@Cantfigurethisout any updates?

By the way check out the new [slim model](https://github.com/intel-isl/OpenBot/blob/master/body/cad/slim_body). Top and bottom both fit on a 220mmx220mm build plate. If this is still too big, check out the [glueable model](https://github.com/intel-isl/OpenBot/blob/master/body/cad/glue_body).

Cantfigurethisout, 2020-09-21:
Haven't gotten back to this.  I saw the slim model, however my build volume is 200x200 (I hear this can be increased with open source firmware on my printer however I haven't gotten to that yet).  Didn't see the glueable model, I will look for that. Thanks

Cantfigurethisout, 2020-09-22:
Ok, I tried again, enabled Verbose logging so if this is too much, I apologize.  Copied error code:

Arduino: 1.8.13 (Windows 10), Board: "Arduino NANO 33 IoT"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\jresp\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\jresp\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\jresp\OneDrive\Documents\Arduino\libraries -fqbn=arduino:samd:nano_33_iot -vid-pid=2341_8057 -ide-version=10813 -build-path C:\Users\jresp\AppData\Local\Temp\arduino_build_869510 -warnings=none -build-cache C:\Users\jresp\AppData\Local\Temp\arduino_cache_836998 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Users\jresp\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.2.1 -prefs=runtime.tools.arduinoOTA-1.2.1.path=C:\Users\jresp\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.2.1 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\jresp\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4 -prefs=runtime.tools.arm-none-eabi-gcc-7-2017q4.path=C:\Users\jresp\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4 -prefs=runtime.tools.bossac.path=C:\Users\jresp\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.7.0-arduino3 -prefs=runtime.tools.bossac-1.7.0-arduino3.path=C:\Users\jresp\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.7.0-arduino3 -prefs=runtime.tools.CMSIS.path=C:\Users\jresp\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0 -prefs=runtime.tools.CMSIS-4.5.0.path=C:\Users\jresp\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0 -prefs=runtime.tools.CMSIS-Atmel.path=C:\Users\jresp\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.2.0 -prefs=runtime.tools.CMSIS-Atmel-1.2.0.path=C:\Users\jresp\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.2.0 -prefs=runtime.tools.openocd.path=C:\Users\jresp\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.10.0-arduino7 -prefs=runtime.tools.openocd-0.10.0-arduino7.path=C:\Users\jresp\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.10.0-arduino7 -verbose C:\Users\jresp\Downloads\OpenBot-master\firmware\openbot_v1_nano\openbot_v1_nano.ino

C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\jresp\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\jresp\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\jresp\OneDrive\Documents\Arduino\libraries -fqbn=arduino:samd:nano_33_iot -vid-pid=2341_8057 -ide-version=10813 -build-path C:\Users\jresp\AppData\Local\Temp\arduino_build_869510 -warnings=none -build-cache C:\Users\jresp\AppData\Local\Temp\arduino_cache_836998 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Users\jresp\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.2.1 -prefs=runtime.tools.arduinoOTA-1.2.1.path=C:\Users\jresp\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.2.1 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\jresp\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4 -prefs=runtime.tools.arm-none-eabi-gcc-7-2017q4.path=C:\Users\jresp\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4 -prefs=runtime.tools.bossac.path=C:\Users\jresp\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.7.0-arduino3 -prefs=runtime.tools.bossac-1.7.0-arduino3.path=C:\Users\jresp\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.7.0-arduino3 -prefs=runtime.tools.CMSIS.path=C:\Users\jresp\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0 -prefs=runtime.tools.CMSIS-4.5.0.path=C:\Users\jresp\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0 -prefs=runtime.tools.CMSIS-Atmel.path=C:\Users\jresp\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.2.0 -prefs=runtime.tools.CMSIS-Atmel-1.2.0.path=C:\Users\jresp\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.2.0 -prefs=runtime.tools.openocd.path=C:\Users\jresp\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.10.0-arduino7 -prefs=runtime.tools.openocd-0.10.0-arduino7.path=C:\Users\jresp\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.10.0-arduino7 -verbose C:\Users\jresp\Downloads\OpenBot-master\firmware\openbot_v1_nano\openbot_v1_nano.ino

Using board 'nano_33_iot' from platform in folder: C:\Users\jresp\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.8.8

Using core 'arduino' from platform in folder: C:\Users\jresp\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.8.8

Detecting libraries used...

"C:\\Users\\jresp\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\7-2017q4/bin/arm-none-eabi-g++" -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10813 -DARDUINO_SAMD_NANO_33_IOT -DARDUINO_ARCH_SAMD -DCRYSTALLESS -D__SAMD21G18A__ -DUSB_VID=0x2341 -DUSB_P [... cut ...]
             ^~~~~~~~~~~

openbot_v1_nano:354:25: error: 'class NewPing' has no member named 'ping_result'; did you mean 'ping_median'?

     distance_cm = sonar.ping_result / US_ROUNDTRIP_CM; // Ping returned in uS, convert to cm.

                         ^~~~~~~~~~~

                         ping_median

Using library NewPing at version 1.9.1 in folder: C:\Users\jresp\OneDrive\Documents\Arduino\libraries\NewPing 

exit status 1

'class NewPing' has no member named 'ping_timer'; did you mean 'ping_trigger'?

thias15, 2020-09-22:
Did you try to disable the sonar, by setting `HAS_SONAR = 0`

Cantfigurethisout, 2020-09-22:
I have a sonar so wanted to keep it.  will try to disable it and see how that works

Cantfigurethisout, 2020-09-22:
It worked, got the wheels spinning.  Thanks for the help.

closed 2020-09-22 by thias15