ob-f/OpenBot #57: When I am training the data, it always display "There are 0 train images and 0 test images"

KevinBuddy, 2020-09-24:


thias15, 2020-09-24:
Please provide more details. What is your folder structure? How did you configure the paths in Jupyter?

KevinBuddy, 2020-09-24:
I have copied all the txt under sensor_data and jpeg files under images into my_openbot_1

KevinBuddy, 2020-09-24:
If I copied the whole directory into my_openbot_1,it will display an error

KevinBuddy, 2020-09-24:
I can not find any document of associate_frames.Is that your private library？

KevinBuddy, 2020-09-24:
Can you give me the information about associate_frames.match_frame_ctrl_cmd?

KevinBuddy, 2020-09-24:
I got it.

thias15, 2020-09-24:
You need to put the complete folders in the my_openbot_1 folder. The structure is train_data --> datasets --> recordings. The associate_frames.py is a python file with the utility functions for matching frames and controls.

thias15, 2020-09-24:
> I got it.

Is the issue solved?

closed 2020-09-26 by thias15

KevinBuddy, 2020-09-30:
thanks, the issue is solved.