ob-f/OpenBot #66: Add automatic stopping based on ultrasonic sensor.

thias15, 2020-09-29:
Add the option to the Arduino code to stop the car within some threshold of an obstacle based on the ultrasonic sensor.

JNK88545, 2020-10-06:
void stop1(){
   ping_time = millis();
  if (distance_cm<=130){
             analogWrite(PIN_PWM1,0);
             analogWrite(PIN_PWM2,0);
             analogWrite(PIN_PWM3,0);
             analogWrite(PIN_PWM4,0);
  }
}
**Can this function be achieved by writing this way? I failed**

thias15, 2020-10-07:
> void stop1(){
> ping_time = millis();
> if (distance_cm<=130){
> analogWrite(PIN_PWM1,0);
> analogWrite(PIN_PWM2,0);
> analogWrite(PIN_PWM3,0);
> analogWrite(PIN_PWM4,0);
> }
> }
> **Can this function be achieved by writing this way? I failed**

I just merged a PR #77 which has this feature.

closed 2020-10-07 by thias15