ob-f/OpenBot #71: In autopolit mode, where is the cmd(indicator) come from?

KevinBuddy, 2020-09-30:


KevinBuddy, 2020-09-30:
It seems like that I have to control the button b and button x on the joy_stick to produce the indicator in the AUTOPOLIT mode.

thias15, 2020-09-30:
It comes from the game controller input. It should be square (-1, left), triangle (0, straight) or circle (1, right). This is only needed if you train a conditional driving policy as shown in the project video.

KevinBuddy, 2020-09-30:
Yes,I know.When train a conditional driving policy,I should input the button b and button x on the game controller.But I don't understand,when autopilot,the indicator will always be 0,how could it fit the training policy.

thias15, 2020-10-01:
The high-level commands need to be provided to the policy during runtime! If there is an intersection with multiple options, the policy needs to know the intent of the user. If you are training a policy that does not need to resolve such situations, e.g. it will just drive around the kitchen block (#44), these commands are not needed.

KevinBuddy, 2020-10-01:
Oh,I see,than you very much.

closed 2020-10-01 by KevinBuddy