ob-f/OpenBot #72: Little warning concerning hyperparameters

Pascal66, 2020-09-30:
I've successfully trained just for testing my implementation, 
this took 15mn with thoses parameters of issues #31 and files  provided by @chilipeppr  
TRAIN_BATCH_SIZE = 10 #128
TEST_BATCH_SIZE = 10 #128

I think it can be usefull to warn users with "low" config, here 32Mo RAM & RTX 2060 6Go Ram
that setting batch size to more that 16 can cause serious problem to use their computer for a little time.

thias15, 2020-09-30:
I believe the default in the Jupyter notebook is right now a batch size of 16. Did you try that? With this I was able to train on a (powerful) laptop using GPU and on a MacBook using CPU. The batch size of 128 requires a powerful workstation (e.g. 128 GB RAM and TitanX GPU). The 100 refers to the number of epochs. This is something else, it determines for how long to train. I will add some notes to the notebook. One option may also be to change the data loader. Right now it buffers the complete dataset if possible. Maybe it can be implemented more memory-friendly at the cost of performance.

Pascal66, 2020-09-30:
I agree, I just explain my own experience, 
Batch of 16 is ok on a notebook, but if people put 128 to have a better train, they can be afraid about what happen 👍 
On s'amuse toujours de toute façon.

thias15, 2020-10-02:
Disclaimer added as part of #75

closed 2020-10-02 by thias15