ob-f/OpenBot #85: AliExpress Arduino Nano link is for ATmega168P Microcontroller instead of 328p

francisduvivier, 2020-10-15:
I was just checking out the Arduino Nano in the body parts and I noticed that the [aliexpress link](https://www.aliexpress.com/item/32866959979.html) is for an ATmega168P while the arduino [firmware documentation](https://github.com/intel-isl/OpenBot/blob/master/firmware/README.md) mentions an ATmega328P microcontroller and the amazon links are also for a ATmega328P.

Will this firmware be compatible with the hardware from the AliExpress link?

thias15, 2020-10-16:
I did not realize that the Aliexpress link points to a ATmega168P Arduino. Never tested such a board, but it should work. The only real difference between 168 and 328 is the amount of memory. The firmware only uses about 25% memory on the 328, so it should be fine.

francisduvivier, 2020-10-16:
Alrighty, I ordered this one some time ago, but it hasn't arrived yet.
When it has and it doesn't work, I'll open a new issue.

closed 2020-10-16 by francisduvivier