ob-f/OpenBot #88: It will be has compile warning when set USE_MEDIAN as 1

yeahydq, 2020-10-29:
It will be has compile warning when set USE_MEDIAN as 1
https://github.com/intel-isl/OpenBot/blob/7ef9820ae0513c6c0994649eedd2553bb39bcbf6/firmware/openbot_v1_nano/openbot_v1_nano.ino#L106
-----------------------------------------------------------------
Error:
/Users/abc/OpenBot/firmware/openbot_v1_nano/openbot_v1_nano.ino: In function 'void loop()':
/Users/abc/OpenBot/firmware/openbot_v1_nano/openbot_v1_nano.ino:237:73: warning: invalid conversion from 'unsigned int*' to 'int*' [-fpermissive]
         distance_estimate = get_median(distance_array, distance_array_sz);
                                                                         ^
/Users/abc/OpenBot/firmware/openbot_v1_nano/openbot_v1_nano.ino:441:5: note:   initializing argument 1 of 'int get_median(int*, int)'
 int get_median(int a[], int sz) {
     ^~~~~~~~~~

thias15, 2020-10-31:
The warning is no problem. I will fix it in the next for.ware update next week.

thias15, 2020-11-04:
Fixed.

closed 2020-11-04 by thias15