ob-f/OpenBot #89: Why can't my car turn?

xyz2k8, 2020-11-03:
Why can't my car turn? Does the control information sent by mobile app use the data of speed sensor?

thias15, 2020-11-03:
No not at this time. Did you try the tests in the Arduino setup via the Serial Monitor?

xyz2k8, 2020-11-04:
I know. I guess the steering control comes from the position of the ROI in the image identified by the neural network, right? I haven't checked the serial port information carefully, because the USB cable was not connected to the computer during the test, which is a bit inconvenient. I think I can send the serial port information to another mobile phone via Bluetooth, or add a window to the app to display these logs in the future.

thias15, 2020-11-04:
Yes, the object detector will produce a bounding box. Based on this bounding box appropriate controls are produced to keep the object centered. If you want to use the autopilot network, you need to train it on your own dataset. I would highly recommend to send controls with the serial monitor first to make sure your car works correctly.

closed 2020-11-04 by thias15

xyz2k8, 2020-11-09:
In the end, it was found that the reason for the inability to turn was that the switch of my phone's automatic rotation was not turned on.