ob-f/OpenBot #90: How to replace the Bluetooth remote controller (PS4)？

xyz2k8, 2020-11-05:
If I train an autonomous driving network by myself, do I have to rely on a Bluetooth remote controller to collect data? Can I use another mobile phone to send control commands via Bluetooth to the mobile phone installed in the car?

thias15, 2020-11-05:
In principle it should work to use another phone as BT controller. But we have not tried this.

xyz2k8, 2020-11-06:
I understand, thank you for your reply.

thias15, 2020-11-06:
Let me know if you were able to get it to work.

38293312, 2020-11-09:
In fact, I prefer AI. You don't need to operate it manually. You can consider replacing the motherboard in the future. Develop voice dialogue or voice control commands!

thias15, 2020-11-09:
Yes voice control is on our list of features to be added, but will still take some time. Feel free to contribute it if you like.

xyz2k8, 2020-11-10:
> Yes voice control is on our list of features to be added, but will still take some time. Feel free to contribute it if you like.

I reviewed the code carefully, trying to find out how to label when collecting the data, so that I could find an alternative.  Does the collected training data contain PS4 control instructions?  I installed a Bluetooth module on the car, which can remotely control the car. If the collected data contains Bluetooth control commands, I can send it to the mobile phone through the USB serial port.  Do you have a better way to replace Bluetooth PS4?

38293312, 2020-11-10:
> 是的，语音控制在我们要添加的功能列表中，但仍需要一些时间。如果您愿意，可以随时贡献它。

Sorry, I don’t know much about the code, but I can give a good idea and mind

thias15, 2020-11-11:
@xyz2k8 yes, you can send the control commands to the phone and then log them inside the SensorService.java class. The data received from the Arduino is in [line228](https://github.com/intel-isl/OpenBot/blob/bcfa1abf72b4af497419399931e3068768a886a7/android/app/src/main/java/org/openbot/SensorService.java#L228). You then need to extract the controls and save them to the control log. Note that the vehicle log is only sent once a second. You will want to send the controls at a higher frequency. I would recommend at least every 30ms. If you do not disable the vehicle log, you will also need a header to know which message you have received from the Arduino. 
```
appendLog(mCtrlLog, SystemClock.elapsedRealtimeNanos() + "," + leftCtrl + "," + rightCtrl);
```

In [line405](https://github.com/intel-isl/OpenBot/blob/bcfa1abf72b4af497419399931e3068768a886a7/android/app/src/main/java/org/openbot/SensorService.java#L405) the controls from the CameraActivity are received. This corresponds to either the controls from a connected BT game controller or from the neural network that's running (you can also log while a network is running). Since you won't have an external controller connected, you can just leave it. But if you want to be sure there is no interference with what you will be writing to the log, you can comment this line.

xyz2k8, 2020-11-11:
@thias15 👍 
Thank you very much for your guidance, this is exactly the solution I expected.  I'll try it this weekend.

thias15, 2020-11-17:
Any luck?

xyz2k8, 2020-11-20:
Sorry, there is no time to deal with this, you can close this issue first, and I will report if there is progress in the future.

thias15, 2021-01-01:
@xyz2k8 there is now a dedicated app that lets you use another phone as remote control for OpenBot. You can download it from the artifacts or compile the controller module in the android folder.