ob-f/OpenBot #93: Test indicator leds

nanotuxi, 2020-11-06:
Building this project is a lot of fun. Thanks for it.
One thing that irritated me at the beginning when the indicator led have been wired and 1st tests were successful everything worked but the leds seemed to be unresponsive. Reading the source code I found the reason was that they only work, if a gamepad controller is attached via bluetooth for training the model. As I do not have a game controller at all (altough I like to play very much ;-) the leds would never work?!?

To see if the wiring of the leds are made corrctly I simply wrote some test code that is exectued in the init phase of the arduino and lets the leds blink once for 1 second by simply adding these lines in the `setup()` of `openbot_v1_nano.ino`:
```
    void setup()
    {
    [... ]
     // test if the led wiring is correct
      digitalWrite(PIN_LED_RB,HIGH);
      delay(1000);
      digitalWrite(PIN_LED_RB,LOW);
      delay(1000);
      digitalWrite(PIN_LED_LB,HIGH);
      delay(1000);
      digitalWrite(PIN_LED_LB,LOW);
      delay(1000);```
    [...]
    }

thias15, 2020-11-06:
Thanks. Will add this to next firmware update.

thias15, 2020-11-06:
This is now added in PR #92.

nanotuxi, 2020-11-07:
Thanks for that.

closed 2020-11-08 by nanotuxi

francisduvivier, 2020-11-08:
Tx!