ob-f/OpenBot #100: Confused about directory structure

zlite, 2020-11-09:
Thanks for the quick response. I am a bit confused about the next step, because the instructions are not clear about where to place the training data you've pulled off your phone. What's the difference between "train_data" and "test_data"? Which one (or both) should the images and sensor data folders go into?

I've set up my directory structure as follows,
[image]

with the images and sensor folders in the train_data folder and nothing in the test_data folder. But that doesn't seem to work.
[image]

Can you add a sample screenshot of the correct directory structure and file locations? The text-tree you've provided so far and the instructions in the Jupyter notebook don't explain this

thias15, 2020-11-10:
The folder structure is explained [here](https://github.com/intel-isl/OpenBot/tree/master/policy#data-collection). Basically, inside each, the train_data and test_data folder, you can make folders for each dataset you collect and give it a name such as "my_openbot_1", "my_openbot_2", etc. Inside these folders, you drop all the zip files from that recording session and unpack them. I can add a screenshot to the repo to make it more clear. 

[image]

The reason for the train_data and test_data folder is to have some data for validation/testing during training that has not been seen. This then gives you an indication during training of how well this policy may transfer. The 

common split is 80% training data and 20% test data.

thias15, 2020-11-10:
Also, you might find issue #31 helpful for getting the training to work.

zlite, 2020-11-10:
Just to be clear, since I can't tell from the screenshot, I should place the *same* Images and Sensor Data folders and files in both the Test and Train folders?  Or should I some runs in Test and *others* in Train so that they have different data to look at, as your note suggests?

zlite, 2020-11-10:
Reading as much as I can, I think the clearer way to put this in the documentation would be with an example, perhaps like this:

1) Collect five data sets (five runs over the same area, pressing record for each one)
2) Copy all those zip files to your PC
3) Unzip them and put four of them in the Train folder, each as an Images (containing all the images) and Sensor Data (containing all the sensor data text files) folder within a folder with the run number
4) Put the remaining run in the Test folder, also within Images and Sensor data folders inside a run number folder. 

This ratio of 80% training/20% testing data is a good rule of thumb for all your data gathering going forward

thias15, 2020-11-11:
The data in the train_data and test_data folders should be different. As you can see in the screenshot in the folders with the different runs (my_openbot_1, my_openbot_2) are just the extracted log folders that come from the OpenBot phone. The folder names correspond to the timestamp of when the log was started (trigger on the controller). I did update the description in the Readme yesterday. Let me know if this is easier to understand.

zlite, 2020-11-11:
Thanks for the explanations (and sorry for being dim ;-) It is clearer now, although I still like the idea of providing an example, as I did above, for those who really want it spelled out.

thias15, 2020-11-11:
Ok, thanks for your feedback, will see how I can incoporate that.

zlite, 2020-11-12:
Having now put the files in a directory structure as instructed, I'm getting this error. There seems to be some confusion: should the Sensor Data folder be *within* the Images folder, as this error suggests, or at the same level as the Images folder, as the documentation says?
[image]

thias15, 2020-11-12:
No, it should be as the documentation says. You are missing one level of directories. Right now you put the extracted zip files directly into `dataset/train_data/`. But there should be subfolders corresponding to the recording sessions. So for example you collect some data for learning to drive around your kitchen block in the morning. Then you create the folder `dataset/train_data/kitchen_morning` and then extract all of the zip files into that folder. Then you may collect more data in the evening, and will put it into `dataset/train_data/kitchen_evening`. By default, you will train on all data. However, you can also specify in the notebook to train only on specific folders. This is useful if you want to train different policies with data collected in different environments (e.g. another subfolder `living_room_morning`). 

Similarly, you might have different openbots, e.g. `my_openbot_1`, `my_openbot_2`, etc. This is the example from the documentation and what I posted a screenshot of above. I hope this makes sense now.

zlite, 2020-11-12:
Sorry for being so dim! I guess I was just confused by the example screenshot you provided, where you have different OpenBot numbers in the Test and Train directories.  Wouldn't it be clearer and simpler to use the same OpenBot number for both, which I presume is the most common case (people only have one)?

Like this:
[image]

thias15, 2020-11-13:
Yes, you are probably right.

closed 2020-11-13 by thias15