ob-f/OpenBot #95: How can I fix the mistake of sharp cornering or turning the corner of the robot while following human?

quockhanh1198, 2020-11-09:
Hello Thias, I ran the model on my Xiaomi Redmi Note 7 phone and it looks pretty ok. But in some cases when human go too fast or a sharp turn, the robot cannot detect people anymore and cannot follow. Can you help me fix these points ? Thank you very much !

thias15, 2020-11-09:
If the person is too close to the robot and makes very sharp turns, the person may be out of the view. In this case there is no way for the robot to recover until the person comes into view again. In addition, the object detector is also not perfect and may not always be able to detect people correctly. One thing you can do to improve detections, is mounting the phone tilted slightly upwards.

closed 2020-11-09 by thias15

quockhanh1198, 2020-11-09:
Thank you for your reply, and please can you tell me why I should tilt the phone up slightly?