ob-f/OpenBot #97: The 3D printing file size is not suitable for the I3 frame printer, ha ha!

38293312, 2020-11-09:
I really like this project and want to DIY one, but my I3 frame 3D printer cannot print OPEBOT files. I have a headache!
I just read the related articles and found that the "speed sensor" has no model. I found it on the Internet and there is no one. Can you tell me an alternative model? Can all modules of this signal be used?
"The slot-shaped through-beam photoelectric, as long as the non-transparent object passes through the slot, it can trigger to output 5V TTL level. Using Schmitt trigger to de-jitter pulse"

thias15, 2020-11-09:
Could you please elaborate why you cannot print? 

I assume the speed sensor is an unrelated issue. What do you mean by model of the speed sensor? It is an optical encoder and the disk attached to the motor shaft, interrupts the optical signal. By counting the interrupts, the number of rotations can be measured.

38293312, 2020-11-09:
> 您能否详细说明为什么无法打印？
> 
> 我认为速度传感器是一个无关紧要的问题。速度传感器的型号是什么意思？它是一个光学编码器，并且光盘固定在电机轴上，可中断光学信号。通过计算中断次数，可以测量转数。

Hahaha, what I mean by unprintable means that the size of the object exceeds the size that the I3 printer can print.
Can the speed sensor not be used?

thias15, 2020-11-09:
You can try the [glueable body](https://github.com/intel-isl/OpenBot/tree/master/body/cad/glue_body). As for the speed sensor, yes it should work. Are you asking about a specific model that's different from the one in the BOM?

38293312, 2020-11-10:
> 您可以尝试可[胶粘的身体](https://github.com/intel-isl/OpenBot/tree/master/body/cad/glue_body)。至于速度传感器，是的，它应该可以工作。您是否要询问与BOM表中的模型不同的特定模型？

Yes, after your reminder, I noticed that there are close-up models in the BOM! Thank you very much, so you can print, I am preparing electronic components, ready to assemble one by hand, thank you again for sharing!

closed 2020-11-11 by thias15