ob-f/OpenBot #98: Errors in running TF

zlite, 2020-11-09:
Despite setting up my conda environment exactly as specified on Windows 10 (conda create -n openbot python=3.7 tensorflow-gpu=2.0.0 notebook=6.1.1 matplotlib=3.3.1 pillow=7.2.0), and running the Jupyter notebook from within that Conda environment, I'm getting TensorFlow errors in the Jupyter notebook:

---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-3-64156d691fe5> in <module>
----> 1 import tensorflow as tf

~\miniconda3\envs\openbot\lib\site-packages\tensorflow\__init__.py in <module>
     96 
     97 # We still need all the names that are toplevel on tensorflow_core
---> 98 from tensorflow_core import *
     99 
    100 # These should not be visible in the main tf module.

~\miniconda3\envs\openbot\lib\site-packages\tensorflow_core\__init__.py in <module>
     43 from . _api.v2 import autograph
     44 from . _api.v2 import bitwise
---> 45 from . _api.v2 import compat
     46 from . _api.v2 import config
     47 from . _api.v2 import data

~\miniconda3\envs\openbot\lib\site-packages\tensorflow_core\_api\v2\compat\__init__.py in <module>
     21 import sys as _sys
     22 
---> 23 from . import v1
     24 from . import v2
     25 from tensorflow.python.compat.compat import forward_compatibility_horizon

~\miniconda3\envs\openbot\lib\site-packages\tensorflow_core\_api\v2\compat\v1\__init__.py in <module>
    662 _current_module = _sys.modules[__name__]
    663 try:
--> 664   from tensorflow_estimator.python.estimator.api._v1 import estimator
    665   _current_module.__path__ = (
    666       [_module_util.get_parent_dir(estimator)] + _current_module.__path__)

~\miniconda3\envs\openbot\lib\site-packages\tensorflow_estimator\__init__.py in <module>
      8 import sys as _sys
      9 
---> 10 from tensorflow_estimator._api.v1 import estimator
     11 
     12 del _print_function

~\miniconda3\envs\openbot\lib\site-packages\tensorflow_estimator\_api\v1\estimator\__init__.py in <module>
      8 import sys as _sys
      9 
---> 10 from tensorflow_estimator._api.v1.estimator import experimental
     11 from tensorflow_estimator._api.v1.estimator import export
     12 from tensorflow_estimator._api.v1.estimator import inputs

~\miniconda3\envs\openbot\lib\site-packages\tensorflow_estimator\_api\v1\estimator\experimental\__init__.py in <module>
      8 import sys as _sys
      9 
---> 10 from tensorflow_estimator.python.estimator.canned.dnn import dnn_logit_fn_builder
     11 from tensorflow_estimator.python.estimator.canned.kmeans import KMeansClustering as KMeans
     12 from tensorflow_estimator.python.estimator.canned.linear import LinearSDCA

~\miniconda3\envs\openbot\lib\site-packages\tensorflow_estimator\python\estimator\canned\dnn.py in <module>
     31 from tensorflow.python.keras.utils import losses_utils
     32 from tensorflow.python.util.tf_export import estimator_export
---> 33 from tensorflow_estimator.python.estimator import estimator
     34 from tensorflow_estimator.python.estimator.canned import head as head_lib
     35 from tensorflow_estimator.python.estimator.canned import optimizers

~\miniconda3\envs\openbot\lib\site-packages\tensorflow_estimator\python\estimator\estimator.py in <module>
     51 from tensorflow_estimator.python.estimator import model_fn as model_fn_lib
     52 from tensorflow_estimator.python.estimator import run_config
---> 53 from tensorflow_estimator.python.estimator import util as estimator_util
     54 from tensorflow_estimator.python.estimator.export import export_lib
     55 from tensorflow_estimator.python.estimator.mode_keys import ModeKeys

~\miniconda3\envs\openbot\lib\site-packages\tensorflow_estimator\python\estimator\util.py in <module>
     73 
     74 
---> 75 class _DatasetInitializerHook(tf.compat.v1.train.SessionRunHook):
     76   """Creates a SessionRunHook that initializes the passed iterator."""
     77 

AttributeError: module 'tensorflow' has no attribute 'compat'

thias15, 2020-11-09:
This has been tested under Windows. Not sure what could have changed. Could you try installing with pip? Alternativly you could also try to install tf 2.1.0 or 2.2.0 if that's available for conda on Windows

zlite, 2020-11-09:
Thanks. Installing via pip ("pip install tensorflow --user") did indeed solve the problem. You might want to add that to the toolchain installation instructions.  

Also, for people unfamilar with Conda, you might want to remind them that when they come back to run it to train again they have to activate their OpenBot environment again (conda activate openbot)

thias15, 2020-11-09:
Yes good point, will add this note. The documentation still needs a lot of work and we are actively working on more features in the background as well. Let me know if you encounter any other issues. 

Contributions are very welcome.

closed 2020-11-09 by thias15

zlite, 2020-11-09:
Thanks for the quick response. I am a bit confused about the next step, because the instructions are not clear about where to place the training data you've pulled off your phone.  What's the difference between "train_data" and "test_data"?  Which one (or both) should the images and sensor data folders go into?

I've set up my directory structure as follows, 
[image]

with the images and sensor folders in the train_data folder and nothing in the test_data folder. But that doesn't seem to work. 
[image]

Can you add a sample screenshot of the correct directory structure and file locations? The text-tree you've provided so far doesn't explain this