ob-f/OpenBot #99: Garbled text in app

zlite, 2020-11-09:
On a Pixel 4, the app shows this Chinese text when connected to OpenBot. Otherwise works normally

[image]

thias15, 2020-11-09:
Is your phone from China? It may be related to the Java locale.

thias15, 2020-11-09:
I have never seen something like this. :) I believe, you are using a cheap Arduino clone with a very long Chinese product name. You can fix this by changing [line 963](https://github.com/intel-isl/OpenBot/blob/08bd925fcdbdd7eadcba8d096ab181a16ec67a1a/android/app/src/main/java/org/openbot/CameraActivity.java#L963) in CameraActivity.java to 
```
connectionSwitchCompat.setText("Connected");
```

Please let me know if that fixes your issue.

zlite, 2020-11-09:
No, it's a standard Google Pixel 4 from the US. Running Android 11. 

Yes, it is a Chinese Arduino clone, although a very nice one (DFRobot Romeo board): https://www.dfrobot.com/romeo.html

zlite, 2020-11-10:
That fix worked, thanks!
[image]

closed 2020-11-10 by thias15