ob-f/OpenBot #105: Issue with Indicator

zlite, 2020-11-12:
The Jupyter Notebook doesn't like the IndicatorLog.txt file

[image]

thias15, 2020-11-13:
Check if the indicatorLog is empty. If there is only the header, you can add `0,0` in the next line. It should add the status of the indicators at the beginning of the log automatically to make sure it's never empty. I will look into this issue. As workaround, you can press the triangle button on the PS4 at the start of the recording, or manually add an entry as described above.

zlite, 2020-11-13:
Yes, that was the problem. One of the indicatorLogs in one of the runs had only the header. I added 0,0 and it now trains

thias15, 2020-11-13:
Great to hear it. I will double-check the Android code and see if I can reproduce and fix this issue.

zlite, 2020-11-13:
I'm now getting errors again, despite checking that all the IndicatorLog.txt entries had a header and at least one entry. Very frustrating. 
[image]

My datasets are [here](https://drive.google.com/drive/folders/14VfBgP-RqKDuHjsf0Dv8rGd025HkR6cn?usp=sharing). Can you see anything obviously wrong with them?

thias15, 2020-11-14:
Some of you ctrl_log files in both train_data and test_data are completely empty. I will revisit the code the make it more robust and produce more meaningful error messages. But I'm not sure how you ended up with empty files. I have collected several hours of data, and not encountered such issues. If you can figure out why you have these empty files, please let me know so I can fix it.

zlite, 2020-11-14:
Wow -- I've got no idea! The only change I made to the Android app was the one you suggested to not display the long Chinese Arduino board name. 

I presume the ctrlLog.txt is tracking my PS4 controller input? But I notice that there are a lot of other zero-length sensor files in some of those runs, too.

I'm not quite sure how to properly exit the Android app when I'm done, so after I press X to stop logging, I just disconnect the phone from the bot and sometimes after that the app freezes. Do you think that a bad exit from the app could cause those zero-length files?

zlite, 2020-11-14:
One other possibility. I changed the Arduino code to use quadrature encoders, and I wonder if the interrupts are disrupting some serial flow at 115k. I also get freezes after a few minutes of driving, like this report:
https://openbot-community.slack.com/archives/C01BRGE07QQ/p1603981059143500

My modified Arduino code is [here](https://github.com/zlite/OpenBot/blob/master/firmware/openbot_new.ino)

Maybe I should be logging at a slower serial rate, like 57k?  It's been a long time since I used classic Atmega 328 Arduinos (as opposed to modern boards, like Teensy or 32u4) so maybe they can't handle 115k serial and interrupts at the same time?

thias15, 2020-11-14:
The best way to exist the app is via the back button. Make sure to wait a little bit after collecting the data before you kill the app, to make sure the data is completely written. 

>I presume the ctrlLog.txt is tracking my PS4 controller input? 

Yes.

>But I notice that there are a lot of other zero-length sensor files in some of those runs, too.

Yes, if the sensors don't exist on your phone some files will be empty. But only the control and indicator logs are currently used.

zlite, 2020-11-14:
Interestingly, on the runs where the ctrlLog size is zero, most of the other sensors are also zero, even though my phone has those sensors and other runs they're not zero. So it definitely looks like a logging issue with the Android app;

[image]

[image]

thias15, 2020-11-15:
Is it possible that the app crashed or was killed in these instances? Because if the log is not closed properly, it may be empty. The file does not get saved after each timestep, but only after the buffer is full or after logging is stopped.

thias15, 2020-11-18:
With further testing I was not able to reproduce. Will close it for now. Please reopen if you have any idea what caused the issue in the logs so I can fix it.

closed 2020-11-18 by thias15