ob-f/OpenBot #107: Cannot test with "c128,128" command!

38293312, 2020-11-13:
After successfully encoding with the latest code, the OLED display is normal. I just made the most basic connection. Now I mainly test whether the 4 motors are normal. Using "Cannot test with "c128,128" command", the motor does not rotate, the serial port software The card does not change on line 10.

[image]
[image]

38293312, 2020-11-13:
I tried both the high and low OE driven by the L298N motor, but it still couldn't rotate.

thias15, 2020-11-13:
It looks like your Arduino is not connected to GND. You need to wire the GND pin from your Arduino to the GND of the L298N.

thias15, 2020-11-13:
Another thing you can check is if the new line terminator is enabled in the serial monitor. By default it should be.

38293312, 2020-11-13:
> 您可以检查的另一件事是在串行监视器中是否启用了新的线路终止器。默认情况下应该是。

The GND and arduino GND of the L298N 12V power supply are connected together, and the test with a multimeter is OK

38293312, 2020-11-13:
thank you very much. The problem is solved!

closed 2020-11-13 by 38293312

thias15, 2020-11-13:
How did you solve the problem in the end?