ob-f/OpenBot #109: Person Following False Positives

mahumada408, 2020-11-13:
Hi All, 

Thanks for the great software!

One thing, I seem to be getting a bunch of false positives where the app falsely labels things as `person`. For example, I've had the bot falsely label a stationary truck as a person. Is there anything I can do? Also, would it be possible to publish out the detection probability? Maybe we could have the motors run when the probability is passed a certain threshold. 

Thanks!

thias15, 2020-11-14:
This is already implemented (see [here](https://github.com/intel-isl/OpenBot/blob/27c299b85f757ef4a9256583564f478239ddcf7e/android/app/src/main/java/org/openbot/NetworkActivity.java#L229)). But the object detectors are not perfect. I'm planning to port yolo-v4 at some point which should lead to more reliable detections.

mahumada408, 2020-11-14:
@thias15 , great! Thanks!

For fun, can I change that threshold and re-build the app? How would I go about doing that? (sorry I have 0 experience with android app development)

thias15, 2020-11-14:
Yes, you can change it [here](https://github.com/intel-isl/OpenBot/blob/27c299b85f757ef4a9256583564f478239ddcf7e/android/app/src/main/java/org/openbot/NetworkActivity.java#L62). Default is 50%. Keep in mind that usually the confidence won't be above 60% or 70%. Also, by tilting the phone upward in the phone mount, you can improve detections since the camera you see you better.

closed 2020-12-07 by thias15