ob-f/OpenBot #116: Android app crashes when logging started

zlite, 2020-11-18:
Your latest version of the Android app (updated two days ago) now crashes immediately every time when logging is started. Tested on Android 11 on Pixel 4

thias15, 2020-11-18:
Thanks for raising this issue. The changes were quite minor, mainly porting everything to the new SDK/API version. I will look into it.

zlite, 2020-11-18:
Might be a permissions thing?

zlite, 2020-11-18:
FWIW, If I just download and install the pre-compiled APK, it works, but asks me for permission to access files, camera, an Arduino, which the one I compiled myself in Android Studio doesn't (perhaps because I load it via developer mode)

thias15, 2020-11-18:
The permission handling is normal. I just finished my meetings and will look into it now. In the meantime, you can download the previous version of the app, for example from release v0.1

thias15, 2020-11-18:
Ok, turns out the file storage permission handline was changed in API30 (Android 11). I have reverted back to API28 for now. This solves the issue for now. I will look for better solution that works well with API30.

closed 2020-11-18 by thias15