ob-f/OpenBot #118: How to charge my phone right on the robot ?

quockhanh1198, 2020-11-20:
Hi Thias, when I run the model with the phone's GPU the performance is very good so I intend to set it up by default, but the phone is quite battery consuming and I am not sure how to maintain the battery of the phone by charging the phone on robot most effective. Can you help me? Thank you so much !

thias15, 2020-11-20:
You could try a cable like [this](https://www.amazon.com/dp/B07CG3MR48). Connect one end to the Arduino and another to a power bank. Another option, if you are good with electronics/soldering is to open a USB cable and connect the 5V lines to the regulator on the L298N.

quockhanh1198, 2020-11-20:
Thank you for your reply

closed 2020-11-23 by thias15

JamesNewton, 2021-03-12:
Many phones will not charge and operate an OTG device at the same time. I made a list of the ones that do. Also of some tablets and phones which have a seperate power jack. 
http://www.massmind.org/techref/io/usb/otg.htm