ob-f/OpenBot #119: How to train the robot to avoid obstacles

Jojo11111, 2020-11-20:
Hello, thanks for your great work on this project. We have problems in training the car to avoid obstacles. We would like to know that if your chairs as obstacles are always placed in the same positions  or placed randomly during each data collection process.

thias15, 2020-11-20:
They are placed randomly during data collection. During test time they are placed in different locations. How much data did you collect? Can you share an image/video of your data collection setup?

Jojo11111, 2020-11-21:
We collect about 6000 images with obstacles by running a python file written by ourselves. We use a game controller to control the car to avoid obstacles and record the datas just like what you did. The result is that the car runs into the obstacles without any tendency to turn around.

thias15, 2020-11-23:
Ok, that's a very small dataset. I don't quite understand: Do you collect data with game controller or with a Python script? Can describe your data collection pipeline and maybe some sample images?

Jojo11111, 2020-11-25:
We collect data with game controller and save the data to matched_frame_ctrl_cmd_processed.txt with a python script

Here are some frames
[image]

[image]

And here is the matched_frame_ctrl_cmd_processed.txt
[image]

According to your previous reply, I think our method is correct, but the dataset is too small. We may try again later and then tell you the result.

Jojo11111, 2020-11-26:
This time we have more images for training and testing but it still doesn't works well.
[image]

we set epoch 60 and here are the training results
[image]
[image]
[image]
[image]

But when the tf model was converted to tflite model, an error occurred.
[image]
We are not sure what caused this error. But when we set epoch 40, the tflite model saved successfully.
Then we used the epoch_40-model for navigation and obstacle avoidance, but the result was discouraging.
Could you give us some suggestions?

thias15, 2020-11-30:
`[Errno 12] Cannot allocate memory` means that you do not have enough RAM memory for the conversion. I guess the memory is already at the limit during training. 

The sudden drop in performance during training is not good. Maybe the learning rate is too high. What hyperparameters do you use? (learning rate, batch size, etc.)?

Jojo11111, 2020-12-07:
[image]

thias15, 2020-12-09:
For training the final model, it is recommended to increase the batch size. You may also try to use another model. The `model.cil_mobile` is a very compact model. You could try for example `model.pilot_net`. It has a higher capacity and may be able to learn your task more easily.

thias15, 2020-12-15:
Closing due to inactivity. Feel free to reopen or reach out on Slack.

closed 2020-12-15 by thias15

Jojo11111, 2020-12-22:
Thanks you very much, we will try it later.