ob-f/OpenBot #124: [Feature] Maintain settings on configuration changes and between app runs

msesma, 2020-12-09:
Current implementation of the app maintains the camera selection on configuration changes, like a device rotation.

This feature proposal if for maintaining all settings not only during configuration changes but also between app runs.

Settings that will be persisted:

- Data Logger, and logging state.
- USB connection speed.
- Drive mode
- Camera
- Vehicle control
- Model
- Device
- Threads

Settings that won't be persisted

- Controller switch (For safety reasons)
- Connection switch (It changes automatically if a device is connected)

thias15, 2020-12-09:
This sounds great! I think the logging state should probably not be persistent either, since logging is usually turned on/off by a user during data collection. Thanks @msesma.

msesma, 2020-12-09:
Sounds reasonable. In that case I´ll make logging state survive configuration changes only.

vkuehn, 2020-12-10:
@msesma there is already some psersistence for the camera switch. please make sure it doesn't conflict with your changes

msesma, 2020-12-10:
@thias15 there is no need to persist camera on configuration changes anymore as it is persisted even between runs. But I will use that to persist log mode instead.

closed 2020-12-11 by thias15