ob-f/OpenBot #143: how to connect ps3 controller to the app?

shark526, 2021-01-09:
This is a awesome project, thanks to all the contributors. I have already built the car success not trying turn to the drive policy training.
I have PS3 controller but could not find any clue on how to connect to the openbot app.
Anyone have such experience on this? Appreciate for your help.

thias15, 2021-01-09:
I'm not 100% sure about PS3 controller. But for a PS4 controller you just pair it with the phone via Bluetooth. There is also a smartphone app now. So if you have another phone you can also use that to control the bot.

shark526, 2021-01-10:
Hi @thias15 , thanks for the reply. Yeah I know the controller app in the repo, but unfortunately I don't have another Android phone.
Connect to ps4 controller must be easier. I had use the ps3 controller connect to Android phone many years ago,  but it required rooted phone and another app.  Wondering how the ps3 controller work with openbot...

thias15, 2021-01-10:
I don't see why the PS3 controller would be a y different from the PS4 controller. Did you try to pair?

shark526, 2021-01-10:
> I don't see why the PS3 controller would be a y different from the PS4 controller. Did you try to pair?

yes，no Bluetooth device could be found on phone after the PS button clicked.
PS3 controller should be the tricky Bluetooth device, I remember for the root phone solution, need connect with a otg wire, use another app to detect the mac address, then pair the Bluetooth...

thias15, 2021-01-11:
Ok, maybe it is easier to just buy a cheap knock-off BT game controller.

closed 2021-01-12 by thias15