ob-f/OpenBot #151: About the policy

wangruoyao, 2021-01-21:
I'm new here and actually don't  know how the training works. Just import the images when I'm driving with trying to achieve some goals such as follow human or along the black line and train them in tensorflow?Do the neural network really know what it is doing?

I'm wondering if I can label the picture manually and train the car more intelligently such as recognize mini traffic light and then do what it is supposed do, and identify a particular people by outlook such as face, clothes, height or something, not just a general human and etc.. 

And even more, if I can train them individually and deploy them on the car?  Like find pictures online to improve the accuracy of recognition and then combine with the manually drive to let it know how to act at the circumstance. Despise the priority(images have two or above different types of labels).

thias15, 2021-01-21:
The autopilot just regresses images to controls. The task needs to be setup in such a way that it is possible to infer the controls only from the image. So for example, if you drive the car to follow a black line and record the data, the network can learn to imitate this behaviour. More complex tasks can also be learned with more data, but the current network only support single-frame predictions. So if the network has to reason about the environment from memory, it won't work. 

The alternative is to build a system. For example like the object detector which is trained offline to detect objects and a large-scale dataset. The predictions are then used to compute controls. You could similarly run multiple networks, e.g. one to detect traffic signs, one to follow the road, one to stop for obstacles, etc. Then you could build a state machine that produces controls taking all those predictions into account.

wangruoyao, 2021-01-22:
Thanks for your reply. Now I know how it works better, it is based on images rather than on labels right？

 I will try it from a simple way like follow a black line and see how it works. 

And then try the second suggestion, which looks like really hard to achieve for me.

closed 2021-01-23 by wangruoyao