ob-f/OpenBot #160: Gerber files request

xebbmw, 2021-02-25:
Hello,

I found this interesting project and I would like to build it as an educational project. 

I am planning to submit the Gerber files to a manufacturer and then solder the parts myself using a hot air station and a solder iron. 

Soldering 0603 SMD parts is doable, but I would prefer instead to solder 0805 SMD. Please provide another version of Gerber files (v2a) where the parts in size 0603 are replaced by 0805 size.

Thanks in advance for your help.

thias15, 2021-02-25:
The gerber files are in the repo. Please refer to the [README](https://github.com/intel-isl/OpenBot/tree/master/body/pcb/README.md) for more info.

xebbmw, 2021-02-25:
> If you prefer I can give you access to the EasyEDA project or export it to a format you can work with. Then you can make any required changes yourself.
Hello, 
Thanks for your help. I would prefer access to the files in Eagle CAD (version 7.50) or KiCAD format. Then I can do the required changes myself. Later I can provide you the new version, maybe others would be interested too.

thias15, 2021-02-25:
That sounds great. Here is the converted file (in KiCAD format). There were a few warnings during the conversion. So please let me know if the attached file works. [
[PCB_OpenBotV2.kicad_pcb.txt](https://github.com/intel-isl/OpenBot/files/6045532/PCB_OpenBotV2.kicad_pcb.txt)
](url)

xebbmw, 2021-02-26:
Thanks, I tried to open the file. Not sure if it is the correct file, because I see the PCB has different components (about 13 resistors, 4 capacitors and an small integrated circuit with 48 pins) and also a different layout. See attached file how it looks in Kicad:
[image]
Also there is another problem, in Kicad I only see pads not the component itself (like resistor for example). It will be difficult in this way to change the component footprint size just by manipulating the pads size.

thias15, 2021-03-02:
Sorry, it seems I uploaded the wrong file by mistake. I have reuploaded. The problem is that I can only export a json file from EasyEDA. I found an online tool to convert to kicad, but it is possible that the components are lost in that process. Here is also the original [json file](https://github.com/intel-isl/OpenBot/files/6071067/DRV8871_Rev2_2021-03-02_19-55-05.json.txt). This is the tool I used [easyeda2kicad](https://wokwi.com/easyeda2kicad). Hope this helps. I have reopened the issue. If this still not works, I can make the changes and provide an updated Gerber file.

xebbmw, 2021-03-03:
Thanks a lot, this time worked just fine! I was able to convert the file and then open it in Kicad. I made the changes for the components C1, C2, LED1, R7, R1, R2, these are now in 0805 format. Here is how it looks right now: 
[image]
I will enlarge just a bit the tracks and then I will generate the Gerber file. I will provide you the files afterwards.

thias15, 2021-03-03:
Thanks @xebbmw. The only thing is that the OpenBot logo and text seem to be messed up by the conversion. But I guess that is not a big deal.

xebbmw, 2021-03-04:
I added the logo that I found on Github, but I could not find the second logo that is in the middle of the board. Also I don't know if there is something else on the other side of the board.
[image]
Please add the second logo so I could insert it also.

thias15, 2021-03-04:
Not sure how to export the icon. You could just find one with a permissive license that looks similar. The icon is just to show the the PCB may become hot during operation and should not be touched. There is nothing on the other side of the board for now.

xebbmw, 2021-03-06:
I added the second logo from the file pcb_2d_v2.png from this repository and then I imported it. This is the new version with a screenshot from the Gerber viewer. 
[image]
If is ok with you, I will send you all the Gerber files so you can add them to the repository.

thias15, 2021-03-08:
Yes, that would be great. Thanks a lot! This will be very helpful for ppl who want to solder by hand.

closed 2021-03-08 by thias15

xebbmw, 2021-03-12:
I attached a zip file with all the Gerber files. Thanks once again for providing me the sources!
[gerber_v2a.zip](https://github.com/intel-isl/OpenBot/files/6127380/gerber_v2a.zip)