ob-f/OpenBot #189: A Issue In Python Controller

Twisted-Fates, 2021-05-07:
Hello!
When I open keyboard-pygame.py or keyboard-click.py，the phone can't connect my computer(Windows 10).

thias15, 2021-05-07:
Can you please elaborate on the problem. Are you seeing some error message? Did you toggle to the "smartphone controller" on the OpenBot app.

Twisted-Fates, 2021-05-07:
Thankyou for your reply!
This is my problem:

Twisted-Fates, 2021-05-08:
[image]
[image]

Twisted-Fates, 2021-05-08:
[image]

Please help me.
I don't know how to deal with it.

thias15, 2021-05-08:
It looks like you executed the `pip install -r requirements.txt` before activating the environment and it installed fine. So try conda deactivate so you are in the base environment where it seems to be installed and then run the script. When you are in the conda environment it seems pygame cannot be found which seems very strange since it is very common. You could also try to install pygame via conda using the command `conda install pygame` after activating the environment.

thias15, 2021-05-08:
When you run the conda install the error message states you don't have the channels for pygame, so you need to try the command I posted above: `conda install -c cogsci pygame`

thias15, 2021-05-08:
When you do not have the environment activated and you see "Waiting for connection", this is the expected behavior, so it seems to be working. However, it seems the devices cannot see each other. If your router has a 2.4Ghz and 5Ghz mode, it could be that the phone and laptop are connected to different versions of your network. So you can try to disable the 5Ghz band on your router.

thias15, 2021-05-08:
Are phone and laptop connected to the same WiFi? Do both have a valid IP address?

Twisted-Fates, 2021-05-09:
My laptop create a Mobile hotspot and phone connect to it.

thias15, 2021-05-09:
@izivkov do you have any intuition if a direct connection via hotspot would work?

izivkov, 2021-05-09:
The connection was meant to be used as two devices on the same network connected to the same router.

I think somebody had a similar problem before. They resolved it by using one phone as a hotspot, and two other phones for the controller and the bot. They had problem if one of the phone was used as the hotspot and running the controller or the bot software on it. I think it has to do with how services are discovered on the network.

thias15, 2021-05-10:
Also, you can try the latest OpenBot app version from master. You can download it from [here](https://github.com/intel-isl/OpenBot/actions/runs/818493537).

Twisted-Fates, 2021-05-10:
[image]
[image]

thias15, 2021-05-10:
Ok, it seems the pygame in the conda channel needs python 3.5. So you need to downgrade your python installation. Or use the Installation outside the conda environment.

thias15, 2021-05-10:
Did you try the app version I liked above outside the conda environment where the pygame installation had worked?

Twisted-Fates, 2021-05-12:
Sorry！I am busy recently.（crying...）

thias15, 2021-05-18:
Closing due to inactivity. If the problem persists, feel free to open again.

closed 2021-05-18 by thias15