ob-f/OpenBot #190: Can't see Video on PC (Windows 10) with python

bender2po, 2021-05-09:
**Describe the bug**
Can't see Video on PC (Windows 10) with python.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'cmd'
2. Type  'python keyboard-pygame.py --video'
3. See error: I cannot see any video stream of the robot.

**Expected behavior**
I am expecting to see a video pop-up and which displays camera-view of the robot.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10

**Smartphone (please complete the following information):**
 - Device: Motorola 5gs plus

thias15, 2021-05-09:
It's not tested on Windows but should work. On the phone are you on the FreeRoam screen? Only on that screen video is implented right now. Does the connection work otherwise, i.e. can you control the bot with the keyboard?

bender2po, 2021-05-09:
Thank you for the response.

I checked for 'Default' and 'FreeRoam' screens. Video stream is not working for both.

I think connection is established because I can see speed and steering values changing on the FreeRoam-screen on the phone, according to the keyboard input.

Currently, I am performing only open-loop testing, i.e., the bot is not connected yet.

thias15, 2021-05-09:
Ok, video should also work open loop. Just the latency is currently still a bit too high to drive. Have you agreed to the permissions for camera and audio? @izivkov do you have any idea what the problem might be?

izivkov, 2021-05-09:
From the screenshot, looks like the python app is disconnected at this time. The last line reads "Received"only, which means the program did not receive any data from the Bot, and this causes the python program to exit. Try restarting the python and establish connection again. 
If no video, try switching between FreeRoam and Default screen, a few times. In FreeRaom screen, should see video.

bender2po, 2021-05-10:
I don't think connection disconnected because I can still control the values (speed, steering on FreeRoam) via keyboard.

I tried as @izivkov suggested. Still no luck. Please see screen-shot below:
[image]

izivkov, 2021-05-10:
Hmm, there should be "VIDEO_SERVER_URL" and "VIDEO_COMMAND" sent to the python program, which I don't see. Did you press the "Controller" button (left most button at the bottom of the FreeRoam screen on the android app)? This should change its icon from a controller to a phone.

izivkov, 2021-05-10:
I can see two reasons why the video is not started:
1. Camera permissions is not set. You can manually check this by going to the app settings, find the OpenBot app, and check permissions. If not set, you can set the camera permission manually from there.
2. The network connection is of type "Nearby", which cannot carry video. This happens if WiFi on the phone is turned off, or you are not logged in to a WiFi network. Make sure both the Phone and the PC are connected on the same WiFi network. As I mention in response to another issue, the Hotspot should not be one of the devices. Please try another hotspot, like another phone, and both the Bot phone and the PC should connect to this hotspot.

izivkov, 2021-05-10:
You also updated to the latest app version, right? The python script works with the latest APK.

bender2po, 2021-05-10:
I am also expecting to see "VIDEO_SERVER_URL" and "VIDEO_COMMAND" in order to satisfy the python's 'if-conditions'. 

1. Camera, Location, Microphone, Storage permissions were given.
2. I tried connecting Computer (python) and Phone (OpenBot) to the router as well as another phone's hot spot. Same result in the both scenarios. That is video is not working. However, I could play with keyboard on computer, and, see speedometer and steering GUI on OpenBot app is changing nicely (ie, no visual delay).
3. I am using 'OpenBot-0.3.1.apk'. Is this the latest one?

izivkov, 2021-05-10:
OK, forget what I said about the network connection type. I deleted this comment. This is not the problem since you are communicating to the python script.
So, just check the following:
1. You are running the latest APK for the bot app.
2. Your camera permission is enables in the setting for the OpenBot app.
3. You are using  a third device for hotspot.

izivkov, 2021-05-10:
You can download the app from here, using the barcode:

https://github.com/intel-isl/OpenBot/tree/master/android

bender2po, 2021-05-10:
The barcode directs me to the version 'OpenBot-0.3.1.apk'. Which is already in use.

izivkov, 2021-05-10:
Yes, I need to investigate a bit more. All the basics are correct and it should work. Maybe you can help by trying with a different phone.

bender2po, 2021-05-10:
I just tried with Oneplus-Nord. Still no luck.

thias15, 2021-05-10:
Can you try the latest OpenBot app version from master? You can download it from [here](https://github.com/intel-isl/OpenBot/actions/runs/818493537).

bender2po, 2021-05-10:
@thias15 It is getting stuck here:
[image]

thias15, 2021-05-10:
Ok, you are almost there now. In principle you should then  see a window with the video stream open. It may take a few seconds. Does it just get stuck there indefinitely? Did you move/rotate the phone in the meantime? This can cause issues since on every rotation it will try to reconnect.

bender2po, 2021-05-10:
Yes, it is stuck indefinitely!!

I locked the phone screen in Potrait in one scenario and Landscape in an another scenario. In both the scenarios the python-script got stuck. (FYI, the phone getting heated up nicely. I guess the phone is doing hard work )

closed 2021-05-10 by bender2po

thias15, 2021-05-10:
Ok, great to hear it! If you figure out at some point what the issue was, it would be very helpful to know for others.

bender2po, 2021-05-10:
Sure, I will post here if I figure out the reason.

moneypi, 2021-05-26:
Hello, I met the same problem, any help?
Thanks!

bender2po, 2021-05-27:
@moneypi I am not expert in this field. This is what worked for me (and a workaround as well):
1. Installed the debug version of android app from here: https://github.com/intel-isl/OpenBot/actions/runs/818493537
2. Replaced some part of 'play_video' method in the VideoPlayer with below snippet
`    def play_video(self):
        if not self.stream:
            print(f'Sream not set')
            return
        print(f'Opening the stream...{self.stream}')
        cap = cv2.VideoCapture(self.stream)

        while(cap.isOpened()):
            ret, img = cap.read()
            #img = cv2.rotate(img, cv2.ROTATE_180)
            cv2.imshow('Video', img)
            cv2.waitKey(1)`
3. (optional) reduce pygame window size from setup_screen
4. Make sure your PC and phone are connected to same WiFi source
5. Open the debug app and select 'Free Roam'. On your PC: run the python code with '--video'. Now, you would see two pop-ups: one for stream and another for pygame control. You should select the pygame window to use keyboard control.
[image]

moneypi, 2021-05-27:
Thanks, my friend fix this for me.
He comment ```andGate.addCondition("camera permission");``` from https://github.com/intel-isl/OpenBot/blob/master/android/app/src/main/java/org/openbot/env/RtspServer.java and rebuild Android APP.
And I got Video Stream.

FYI, my phone is Redmi K30 5G/MIUI 12.1.8.

izivkov, 2021-05-27:
Hmm, you need camera permissions. You have bypassed it for now, but can
cause problems in the future.

The first think I would check is if the app has camera permissions from the
setup.

If not, you can set it manually. But I would uninstall the app, build it
again, and it should ask you to grant permission for the camera.

If not, you maybe running a pretty old version of Android. What is your
Android version.

moneypi, 2021-05-27:
Android Version 11
And，when running the app first time, I got the question: Take photos and record videos. And I select "Allow only while using the app".

I am not familiar with Android, maybe that's the reason.

izivkov, 2021-05-27:
Thanks, I'll check if the code has a problem with android 11.

moneypi, 2021-05-28:
My friend told me that is the problem of MIUI, which is different with the standard Android.

BaoAn21, 2025-04-03:
I see that this is a closed issue, but I face this and can not make it work, could anyone help me

[image]
I got to this point, and pygame was just freeze (you can see my mouse is loading) I need to force quit it because it was not run.
I use Samsung A06

BaoAn21, 2025-04-03:
I see that this is closed issue, but I face this and can not make it work.

[image]
I got to this point, and pygame was just freeze, I need to Force Quit the app, no video streaming happen
I use Samsung A06,
Could anyone help me? Thank you in advance