ob-f/OpenBot #191: The Policy Training Web have some problem

Twisted-Fates, 2021-05-10:
I can't open this website:

`http://localhost:8000/#/uploaded`

thias15, 2021-05-10:
Do you already have any logs uploaded? Could you provide a screenshot or error message? @sanyatuning do you have any idea?

Twisted-Fates, 2021-05-12:
[image]

thias15, 2021-05-12:
Are you sure the Python server is running? Are you able to navigate to http://localhost:8000? Did you already record any logs on the robot that have been uploaded?

thias15, 2021-05-18:
Please check the python output for any errors. Sometimes there are faulty datasets. If yes, delete the datasets in the uploaded folder and try again.

closed 2021-05-18 by thias15