ob-f/OpenBot #213: Add to documentation the steps of setting up autonomous navigation through obstacles

ajohansson-git, 2021-07-30:
The OpenBot Youtube video (at 3:00 and 3:17) demonstrates autonomous navigation, including through randomly placed obstacles.

Can I request a beginner-level how-to for setting this up (including training the model or using your pre-trained model, etc.)?

The arxiv paper's methodology description is perhaps not simple enough to reproduce (especially for those who are less experienced with Tensorflow, etc.).

Thank you.

thias15, 2021-08-16:
We will look into this, but it may take some time. Can you see if the latest jupyter notebook on master makes it easier?

thias15, 2022-03-15:
Please check out the [policy learning videos](https://github.com/isl-org/OpenBot/tree/master/body/lite#step-by-step-video-guides).

closed 2022-03-15 by thias15