ob-f/OpenBot #215: video stream problem: pps_id 1388 out of range

thomasfermi, 2021-08-16:
**Describe the bug**
When I run `python keyboard-pygame.py --video`, I do not see a video but rather the following error messages in the terminal

```bash
Starting video...
Opening the stream...
[h264 @ 000002203bf6e0c0] pps_id 1388 out of range
[h264 @ 000002203bf6e0c0] pps_id 1388 out of range
[h264 @ 000002203bf6e0c0] decode_slice_header error
[h264 @ 000002203bf6e0c0] no frame!
Checking the stream...
[h264 @ 000002203c03eb00] pps_id 1388 out of range
[h264 @ 000002203c03eb00] decode_slice_header error
[h264 @ 000002203c03eb00] no frame!
Can't read stream
```

**To Reproduce**
On Windows 10, do
```bash
conda create -n openbot python=3.7
conda activate openbot
cd python
conda install pip
pip install -r requirements.txt
python keyboard-pygame.py --video
```

(I also tried to create an environment from the `policy` folder (without tensorflow, since I do not need that) first and then install the requirements from the `python` folder. Makes no difference)

**Expected behavior**
Video Stream should be visible

**Desktop:**
 - OS: Windows 10

**Smartphone:**
 - Device: Pixel 4a
 - Android 11

**Additional context**
I can view the stream in VLC media player. I also tried adding the following snippet to the `VideoPlayer` class in `keyboard-pygame.py` (right after `print("Can't read stream")`; depends on [python-vlc](https://pypi.org/project/python-vlc/))
```python
 import vlc
 player=vlc.MediaPlayer(self.stream)
 player.play()
```
That leads to the video stream being played. But of course this is not a proper solution.

thomasfermi, 2021-08-16:
Update: Just tried the same thing with an Ubuntu machine and I get the same error. So maybe something is wrong with the way I set up my conda environment? Or is it the Android app?

thias15, 2021-08-16:
@izivkov could you look into this?

izivkov, 2021-08-17:
I will take a look at this.

thomasfermi, 2021-08-17:
Hi @izivkov , thanks for taking a look! I tried with the latest [master build](https://github.com/isl-org/OpenBot/actions/runs/1137164158) but got the same results. Later today I will try with another phone, currently I only have mine.

By the way: When I connect with VLC, the video lags by roughly 2 seconds. Is that also the lag you experience?

izivkov, 2021-08-17:
> Hi @izivkov , thanks for taking a look! I tried with the latest [master build](https://github.com/isl-org/OpenBot/actions/runs/1137164158) but got the same results. Later today I will try with another phone, currently I only have mine.
> 

OK, trying another phone will narrow the problem. My best guess is that unlike VLC, python's `opencv` does not support some codec in some phones. I read there is a way to force a specific codec on the python side, like MP4. 

> By the way: When I connect with VLC, the video lags by roughly 2 seconds. Is that also the lag you experience?

Yes, there is a delay in the video. This is typical for RTSP protocol. The Android app also supports WebRTC, which is much better, but it only works with the Android controller app. I have not been able to figure out how to get it to work with Python nativity. It is also a serverless version WebRTC (to make it easier to setup with the controller app), so will not work with a standard WebRTC client, like a Node/Web Browser. Maybe another feature is to add `standard` WebRTC.

thomasfermi, 2021-08-17:
I just tried with a Huawei phone. I get until the point when the "Stream resolution" is being printed to console:
```bash
Starting video...
Opening the stream...
Checking the stream...
Stream resolution: (720, 720, 3)
```
Then the python application freezes. I tried with the master build of the app, and the regular version.

Regarding RTSP and WebRTC: Would be really nice to have WebRTC work with the python app, since 2s delay is really a problem...

izivkov, 2021-08-17:
OK, thanks for trying. So the phone does make some difference. One basic think to try is to check the permissions on the OpenBot app. Permissions should be set when the app if first run, but just to be sure: From your phone settings -> "Apps and notifications" -> "OpenBot" app -> Permissions, and make sure all permissions are enabled, especially "Camera".

I remember for Huawei phone in particular, somebody had a problem reading the permissions from the app, even if they are set correctly. He fixed the issue by commenting line 60 in RtspServer class in the app.

```// andGate.addCondition("camera permission");```

You may want to try that.

izivkov, 2021-08-18:
Nevermind, ignore my previous comment. I forgot you could stream to VLC.

izivkov, 2021-08-20:
OK, since `VLC` seems to work better than `opencv`, I will replace the opencv in the python script with this:

```https://pypi.org/project/python-vlc/```

Now, python/pygame is not really my forte, so if somebody here has a burning desire to take a crack at it, please go ahead.

izivkov, 2021-08-20:
Just noticed, for the above to work, a VLC player has to be installed on the machine. So this is not ideal. Any alternative ideas will be appreciated.

thomasfermi, 2021-08-25:
Yeah, you are right, `python-vlc` works, but depends on a VLC installation. Maybe it is better to spend time on getting the WebRTC connection to work, rather than improving the current RTSP connection (which is laggy).

Sadly I do not have any knowledge in this domain... Once I find some time, I will try to learn a bit about WebRTC

izivkov, 2021-08-25:
@Mention ***@***.***>  Ok, I'm a little busy these days, so
cannot work on this at the moment. If you decide to work on it, I can be of
assistance.

izivkov, 2021-09-01:
@thias15 , @thomasfermi
I am looking at how to implement WebRTC for the desktop. AFAIK, there is no python library to display the WebRTC stream, similar to RTSP. So, the video can be displayed in the browser, with some JS. So, here is my approach:
- Write a small Python program to connect to the bot app as it does now. This way there is not change the to bot app.
- The Python app will act as a web server to the web browser and the JavaScript. It will also proxy the commands from the bot to the JS and browser.  
- The JS code will handle all WebRTC protocol, and will show the video in the browser.
I makes sense the commands to the bit to be sent from the browser. Also, I have not fugured a way for the Python code to act as a keyboard handler and a web server at the same time, but maybe this is possible. This way we only have one window on focus. 
- There may need to be some UI controls  in addition the the video element to drive the robot.
- In this way, this essentially becomes a JS project, because all the logic will reside on the browser. I like to get your opinion of this is something worth doing, or if you have any other suggestions.

thomasfermi, 2021-09-05:
Hey @izivkov ! I am happy to give my personal opinion :)

From my perspective the promise of the python app is the following: Yes, you can control the bot via keyboard, but what is even better is that the python app is an entry point for personal projects. The code should give easy access to the robot's sensors (including video) as well as its actuators. Python is a great language for beginners to start playing with the bot.

If there is no suitable library for grabbing the WebRTC stream in python, this is a big problem of course. I found [this library](https://github.com/aiortc/aiortc) but probably you already checked that out. 

The idea with JS code sounds promising! I am not sure however what the purpose of the python app would be then. Maybe one should do everything in JS in this case? A lot of people know JS, so the app could still be an entry point for beginners to build personal projects. I do not know much JS, but I would certainly play around with such a JS app.

izivkov, 2021-09-06:
@thomasfermi , thanks for your input. I looked at `aiortc` at one time, but I thought it had too many OS dependencies. However, at a closer look, this may not be a big issue, if at all. Looks promising, and I will attempt to integrate it in the bot's python code.

thomasfermi, 2021-09-06:
Hey @izivkov , that sounds awesome :) Looking forward to it!

izivkov, 2021-09-21:
Unfortunately, I run into problems interfacing python WebRTC to the open bot android app. The two can exchange information, but they fail to negotiate WenRTC capabilities. I think this is due to some incompatibilities between the implementation of `aiortc` and and Google's Android implementation `https://webrtc.org/`. 
  
Second problem is playing the video in python. All the `aiortc` examples I have seen use  python as the source of the video, and a web browser as the display. In our case we want to use android app as the source, and python to display the video. 

In any case, if somebody is interested to investigate further, I have put the related code on my GitHub repository here:

https://github.com/izivkov/OpenBot/tree/ivo/python-webrtc

thomasfermi, 2021-09-29:
Hi @izivkov , thanks for looking into it. Too bad that there is an incompatibility.

thias15, 2021-12-21:
@thomasfermi, please try out the new node-js server by @izivkov

closed 2021-12-21 by thias15