ob-f/OpenBot #216: Taobao purchase links on CN version of `build your robot body` are unable to use

AmagiDDmxh, 2021-08-16:
**Describe the bug**

Links in the section of "料清单" are unable to open and use, which simply redirect to Taobao home page all the time whether I've logged in or not

PAGE LINK: https://github.com/isl-org/OpenBot/blob/master/body/README_CN.md

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'https://github.com/isl-org/OpenBot/blob/master/body/README_CN.md'
2. Click on any '淘宝购买' in “料清单” section
3. Open up Taobao link but non of any item detail page was shown

**Expected behavior**
Should open up a detailed HW item page

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Mac OS
 - Browser Chrome
 - Version 92.0.4515.131

**Additional context**

Haven't found anything useful, the share link format of Taobao seems to change every year? Let me try to figure out this!

thias15, 2021-08-16:
Can you fix it @leegang or @AmagiDDmxh and send a PR. In general the Chinese documentation is very updated and could really use an overhaul across the complete repo.

thias15, 2021-09-10:
Anyone who can updated the Chinese links and documentation?

closed 2021-12-21 by thias15