ob-f/OpenBot #230: .github/workflows/traffic.yml: All jobs have failed

vkuehn, 2021-10-18:
**Describe the bug**
after having cloned that repository for while suddenly I get that error

.github/workflows/traffic.yml: All jobs have failed

looking at the yml it looks like it is getting statistics about the repo from another repo.

**To Reproduce**
just fork the repo

**Expected behavior**
no issues after forking the repo.

Multiple solutions are possible here. However forking is a part of the github standard flow so any solution to it is greatly appreciated.

**Screenshots**
not applicable

**Desktop (please complete the following information):**
github repo issue

**Smartphone (please complete the following information):**
not needed in that context

**Additional context**
already started the discussion in slack about it

closed 2021-11-12 by thias15

thias15, 2021-11-12:
There are two possible solutions:
1) Go to your fork, e.g. in your case https://github.com/vkuehn/OpenBot/settings/actions and disable all actions or filter out the traffic.yml.
2) Delete the traffic.yml from the workflow folder in your fork: https://github.com/vkuehn/OpenBot/blob/master/.github/workflows/traffic.yml

vkuehn, 2021-11-15:
than I would be never able to contribute again...right ?