ob-f/OpenBot #233: Add button to web interface to manually trigger data preprocessing

thias15, 2021-11-10:
Add button to the uploaded tab which will scan the upload folder for items that have been added manually and preprocess them.

thias15, 2021-12-21:
@sanyatuning

sanyatuning, 2021-12-21:
@thias15 a simple page reload will rescan the uploaded dir.
Do you want a "redo matching" to this menu?
[image]

Or something else?

thias15, 2022-01-31:
Redo matching in this menu would be great.

thias15, 2022-05-11:
@sanyatuning what would actually be great is:
(1) Add an option to redo matching for datasets to the dropdown. This can be used if users added data manually. 
(2) Run redo matching when dataset is moved since the file paths in the metadata won't be correct anymore. Running the matching code again will also updated the filepath for each data sample.

thias15, 2022-10-13:
Hi @sanyatuning. Just wanted to follow up if you would still be interested to work on this?